robotology/robometry #1: Kickoff meeting on telemetry development

traversaro, 2020-11-13:
Hi everyone, I scheduled a meeting on Monday 2020/11/16 at 3 PM CET to kickoff the activities on the development of this libraries, mainly to clarify the objectives and users stories we want to address to @AlexAntn @Nicogene . 

People invited: 
@AlexAntn 
@Nicogene 
@S-Dafarra 
@vtikha  
@traversaro 
 
(If any of you want to reschedule for any reason, please let me know, thanks!) 

@drdanz if you like, feel free to join, thanks! 

cc @pattacini @DanielePucci @lornat75 @maggia80

pattacini, 2020-11-13:
Thanks @traversaro 

Could you please forward the invitation anyway?
I'll try to join, at least at the beginning, if possible.

traversaro, 2020-11-13:
Done!

traversaro, 2020-11-23:
Weekly meetings have been scheduled, and for all the planned activities we opened issues in this repo.

closed 2020-11-23 by traversaro