robotology/robometry #4: Prepare a simple example to save a vector and a matrix with matio-cpp

traversaro, 2020-11-16:


S-Dafarra, 2020-11-16:
This is a first example of usage (beside the ones in the README or in the tests): https://github.com/dic-iit/bipedal-locomotion-framework/blob/535fe059fe555cda964f9db83eac269578b43121/utilities/MasImuTest/src/MasImuTest.cpp#L1141-L1167

The output file contains a struct named ``settings`` that is as follows
[image]

In the same file, I also save a ``StructArray`` to store the information of several info I saved from the robot: 
https://github.com/dic-iit/bipedal-locomotion-framework/blob/535fe059fe555cda964f9db83eac269578b43121/utilities/MasImuTest/src/MasImuTest.cpp#L763-L883

The output is like
[image]

traversaro, 2020-11-20:
If we want to also do a bit more advanced case, we can also look into saving a struct with this format on matio: https://github.com/robotology-playground/yarp-telemetry/issues/8 .

AlexAntn, 2020-11-20:
> If we want to also do a bit more advanced case, we can also look into saving a struct with this format on matio: #8 .

I think it makes more sense, considering that @S-Dafarra already provided examples on how to write both a vector and matrix. I tested a minimal example (using the instructions provided on the dic/matio-cpp github page) and everything compiled and ran smoothly, and the installation of the library was fairly easy and fast.

PR #21 (2020-12-07): Add matioCpp examples
This pull-request introduces 2 simple examples of matioCpp use (for vectors and matrices), and an example implementation of the structure for saving timeseries in `.mat` files.

This fixes #4 and fixes #8 .
closed 2020-12-07 by Nicogene