robotology/robometry #5: Brainstorming on the initial prototype

traversaro, 2020-11-16:


AlexAntn, 2020-12-17:
I think for the next PI review we can have the pipeline ready for one sensor (so, read from sensor -> write to circular buffer -> read from buffer and write file).

This would run directly on the machine collecting this data, so the file is written there. 

We could potentially have yarp in the middle: 
- read from sensor -> write to circular buffer -> read from buffer and write to port -> read from port and write mat file. 
- This would allows us also to start working towards collecting all telemetry in a single machine

Nicogene, 2020-12-17:
>We could potentially have yarp in the middle:
    - read from sensor -> write to circular buffer -> read from buffer and write to port -> read from port and write mat file.
    - This would allows us also to start working towards collecting all telemetry in a single machine

@AlexAntn do you mean that the sensor is attached to a machine and the dump is done in a different machine?

I'd remain as much as local as possible, I think the less the network is involved, the better it is and maybe it could be good involving some zero-copy communication in order to pass the memory from the buffer to the thread that has to dump the mat file.

Obviously, it introduces drawbacks, we have to lock the memory and maybe we need the double buffer mechanism @AlexAntn was talking about the last meeting.
We may start making it simple and copy, but we have to keep in mind that this has some scalability issues

AlexAntn, 2020-12-18:
I thought about the yarp connection in order to centralize the actual logs into a single computer. So basically we would have two levels to consider:

- Getting the data from the sensors, writing it to the circular buffer and then reading from the circular buffer
  - This part should be, as Nicolo said, as zero-copy as possible in order not to lose any data
- Getting the data from the second buffer (if we use it - I think it makes sense) and writing the file
  - In this second step we could put the yarp port and connect all telemetry to a centralized module that writes the mat files

This way it would also be easy to access the files (they would all be in the same machine, in the same folder), but it would require this central module to be run separately

traversaro, 2021-01-25:
The first MVP-1 is now available as part of the repo, and on the top of that we are improving it.

closed 2021-01-25 by traversaro