robotology/robometry #8: Understanding how to encode timeseries in .mat file

traversaro, 2020-11-16:


S-Dafarra, 2020-11-20:
Matlab scopes can log the input data as "Structures with Time": https://it.mathworks.com/help/simulink/ug/data-format-for-logged-simulation-data.html#bs40i1i-7
In short, the output is a single struct per signal. It is as follows:
```
- time
- signals
      |
      -- values
      -- dimensions
      -- label
      -- blockName
      -- inReferencedModel
```
``values``, which can be accessed as the first field of the struct ``signals`` contains all the values concatenated.

traversaro, 2020-11-20:
Thanks for the reference @S-Dafarra ! Do you think such a structure can be reproduced with `matio-cpp`? cc @Nicogene @AlexAntn

S-Dafarra, 2020-11-20:
> Thanks for the reference @S-Dafarra ! Do you think such a structure can be reproduced with `matio-cpp`? cc @Nicogene @AlexAntn

Yes, absolutely. I can give support for that

AlexAntn, 2020-11-20:
trying to understand how this structure works, lets say we get 3 values at times 1s, 2s and 3s. In this case, we would have:

```
- time = [1, 2, 3]
- signals
      |
      -- values = [value[T=1]. value[T=2], value[T=3]]
      -- dimensions = dim(value) * length(T) 
      -- label 
      -- blockName 
      -- inReferencedModel
```

I did not really understand the last two elements. for `blockName` I assume it's the name of the simulink block/connection, so should/would this be a vector if we are storing data from more than one block?

traversaro, 2020-11-20:
> I did not really understand the last two elements. for blockName I assume it's the name of the simulink block/connection, so should/would this be a vector if we are storing data from more than one block?

I guess probably in our case we can ignore these two parameters, but probably @S-Dafarra can provide more input on this.

S-Dafarra, 2020-11-20:
> I did not really understand the last two elements. for `blockName` I assume it's the name of the simulink block/connection

You assumed correctly. I just reported what Matlab does in this case, that may be a possible strategy. I guess that, in our case we can substitute the fields like ``blockName`` and ``inReferencedModel`` with others containing info relevant to the logged variable/s.

drdanz, 2020-11-23:
> - time = [1, 2, 3]
> - signals
>       |
>       -- values = [value[T=1]. value[T=2], value[T=3]]
> [...]

What if we have 2 different signals with different timestamps? (i.e time = [0.5, 1.5, 3])

Do we have 2 different `time` vectors, or one single merged (i.e. [0.5, 1, 1.5, 2, 3])) that we have somehow to match with the signals?

S-Dafarra, 2020-11-23:
> What if we have 2 different signals with different timestamps? (i.e time = [0.5, 1.5, 3])
> 
> Do we have 2 different `time` vectors, or one single merged (i.e. [0.5, 1, 1.5, 2, 3])) that we have somehow to match with the signals?

We can also have two separate structs, each one with its own time vector. Indeed, every scope in Simulink save a different struct of this kind. If a scope is displaying multiple inputs, ``signals`` is an array, where an element corresponds to a signal. In fact, in this latter case, the ``time`` vector would be the same for all the signals. Here an example (the fields appear a little different from what I found in the documentation): 
[image]

For example, in order to access the ``left_arm`` data we would need to do (in Matlab)
```
out.JointsPosition.signals(2).values
```
So we may have all the signals sharing the same time vector in the same struct.
 In any case, this is just an example. I think that we are free to choose the representation that best suits our needs, also in terms of simplicity, or access to the data. For example, in this case, in order to access the ``left_arm`` data it would have been preferable to do
```
out.JointsPosition.signals('left_arm').values
```
that would be possible if ``signals`` was a struct itself instead of a struct array.

AlexAntn, 2020-12-03:
Regarding the discussion from the last meeting, as @S-Dafarra mentioned, the names of the signals are lost inside the file, but all the data is there and structured properly!

I implemented the change we discussed in the meeting (using a Struct instead of a StructArray) and it works perfectly. I checked in Octave, and we can access the values of the joints by specifying `output.signals.left_arm.joints`, as discussed.

I will commit it now and start the pull request

PR #21 (2020-12-07): Add matioCpp examples
This pull-request introduces 2 simple examples of matioCpp use (for vectors and matrices), and an example implementation of the structure for saving timeseries in `.mat` files.

This fixes #4 and fixes #8 .
closed 2020-12-07 by Nicogene