robotology/robometry #9: Understand how to ensure mutual exclusion of the buffer

traversaro, 2020-11-16:


traversaro, 2021-02-03:
This starts to be a problem given https://github.com/robotology-playground/yarp-telemetry/pull/55/files that creates a thread that is accessing the buffer on the top of the thread that is logging the data. Mutual access needs to be added, understanting what boost::circular_buffer is doing w.r.t. to that inside, and then understanding a proper stategy to add eventual mutual access.

AlexAntn, 2021-02-19:
Checking the documentation on the circular buffer, it seems that reading is thread safe, but not writing. As such, we should take care to lock and unlock the buffer when there is such an access.

traversaro, 2021-02-22:
Possible ideas:
* Check what yarpdatadumper does 
* To speed up the buffer save to file and minimize the time spent in locking the buffer, we could think of copying the buffer in a different buffer that then is saved to file

Nicogene, 2021-03-01:
Not sure if this is related to the mutual exclusion, but the unit tests seem to fail in a non-deterministic way under macOs.
See https://github.com/robotology-playground/yarp-telemetry/runs/1989172341?check_suite_focus=true

Nicogene, 2021-03-12:
Somehow related also to #94

Nicogene, 2021-03-12:
Probably we have to lock the push_back calls (see https://github.com/robotology-playground/yarp-telemetry/pull/93#issuecomment-797651989), but maybe we can reduce the lock in the save(i.e. unlock as soon as we copied the data to save)

Nicogene, 2021-03-17:
From https://github.com/robotology-playground/yarp-telemetry/issues/87#issuecomment-800918428 it is clear how the not correct mutual exclusion lead to or incorrect number of sample or dumping garbage(4 samples over 11k)

PR #105 (2021-03-24): BufferManager: fix multithreading push_back
It introduces the `BufferInfo` struct that holds the Buffer, the dimensions, and the mutex of the respective buffer. The push_back now lock the mutex of the buffer we are writing to it, keeping the others free.

It fixes #9 

cc @S-Dafarra
commit 76bd8f88 (2021-03-24): BufferManager: lock push_back using the BufferInfo structure

It should fix #9
closed 2021-03-24 by Nicogene