robotology/robometry #16: Define a user story for the telemetry system

S-Dafarra, 2020-11-23:
I tried to think about the following scenarios. @pattacini let me know if this is ok. Open to discussions. 
## Complex Demo
A student runs a complex demo on the robot using Wi-Fi involving several parallel modules communicating with each other and with the robot. The robot experiences a major failure. After collecting some robot parts from the ground, the student goes to the robot workstation to retrieve a log file/folder. He loads the logged data on an application, investigating the possible reasons behind the failure:
 1. there was a jump in the desired values coming from a joystick
 2. the controller was in a wrong state
 3. the robot had a mechanical/electrical issue.

The student checks the data discovering that both 2. and 3. occurred, Since the electrical failure occurred more than 20ms after the controller error, the student feels a little guilty.

## Occasional failure
The robot is enduring an intensive test session. After about one hour and thousands of messages sent from the boards, one of the robot hands is no more responsive. The student retrieves the logged data, determines the instant in which he lost connection by visualizing the joint data and the recording of a synched video camera, and find the corresponding error coming from the board.

## Experiments on the robot for a paper
At 5.45 AM, the student realizes he needed some additional plots for the paper he has to submit at 9.00 AM. In particular, he needed to plot a set of IMU data he was not planning to use before. The student loads again the data he previously backed up, plots the IMU data he was hoping to find, and take a coffee.

## Remote debugging
A new iCub user from Japan runs the red ball demo on the brand new robot in his lab. Something is weird and the robot misses the ball every time. He already has the logged data from that single run and he sends it to Genova to have support.

DanielePucci, 2020-12-15:
Nice notes from Boston Dynamics: 

> We are very data driven [...] **we leverage power tools for logging, analysis and visualisation** in order to drive very high experimental efficiency, so we are able to quickly identify the root causes of the problems [...] and **that's an important key to our success in Atlas** 
 
Video link at 15:27: https://slideslive.com/38946802/boston-dynamics?fbclid=IwAR1fZHJj7k7sCOQ5O2mCvPWshi-Oc1whlvYhiReTD35JJve20455HR4gJKA

cc:

$dic-iit/sw-dev
$hsp-iit/sw-dev
$edpr-iit/sw-dev
@icub-tech-iit/telemetry

DanielePucci, 2020-12-15:
(do not know if the bot is working, so I do)

CC @pattacini @diegoferigo @S-Dafarra @gabrielenava @Yeshasvitvs @lornat75 @randaz81 @drdanz @elandini84 @ChiaraBartolozzi @arrenglover

traversaro, 2020-12-15:
Hi @DanielePucci, this repo is on robotology-playground for now, so the external collaborator notifications are not active. 

FYI @AlexAntn @Nicogene the presentation https://github.com/robotology-playground/yarp-telemetry/issues/16#issuecomment-745150983 is quite interesting.

traversaro, 2020-12-15:
> Hi @DanielePucci, this repo is on robotology-playground for now, so the external collaborator notifications are not active.

It worked on @pattacini comment, because it was done when the issue was on icub-tech-iit .

traversaro, 2021-01-25:
A first MVP have been presented (see https://github.com/robotology-playground/yarp-telemetry/issues/25), and even if it does not realize all the points (see https://github.com/robotology-playground/yarp-telemetry/issues/37) I think we can close this issue as the specific aspects of the user story will be tracked by separate issues.

closed 2021-01-25 by traversaro