robotology/robometry #23: Fix/investigate a way to express the payload of a single Record

Nicogene, 2020-12-17:
In #22 it has been added just as a sum of the sizeof of the members of `Record` but this does not work with the containers(e.g. see http://cpp.sh/5utpi)

This kind of info may be useful for defining a RAM cap occupied by the circular buffer and then its maximum length in function of which kind of data is carrying

traversaro, 2021-01-05:
I guess the only way of computing the memory used and owned by a C++ object of an arbitrary type is to have a templated function, something like (probably this is not valid C++, but I think you can get the idea): 

~~~
template<typename T>   // primary template
size_t getUsedMemory(const T& instance) 
{
    return sizeof(T);
}

size_t getUsedMemory<std::vector<double>>(const T& instance) 
{
    return sizeof(instance) + sizeof(double)*instance.capacity();
}
~~~

and then specialization need to be provided for all the types we want to support.

Nicogene, 2021-01-08:
In this sense, I was thinking about refactoring the `Record` API.
Now the `m_datum` is of type T
https://github.com/robotology-playground/yarp-telemetry/blob/9562ab244fe897ef3dc2bcdd831e791864e890da/src/libYARP_telemetry/src/yarp/telemetry/Record.h#L20
I would make something span like, that in case of scalar record would have `size = 1`.

In this way the `getUsedMemory` would be `sizeof(span<T>) + sizeof(T)*m_datum.size()`

We can't use the the `std::span` but we can use it something like this (https://github.com/boostorg/beast/blob/develop/include/boost/beast/core/span.hpp) or implement our own `telemetry::Span`

What do you think about it?

cc @traversaro @AlexAntn @S-Dafarra @drdanz

S-Dafarra, 2021-01-08:
> In this sense, I was thinking about refactoring the `Record` API.
> Now the `m_datum` is of type T
> https://github.com/robotology-playground/yarp-telemetry/blob/9562ab244fe897ef3dc2bcdd831e791864e890da/src/libYARP_telemetry/src/yarp/telemetry/Record.h#L20
> 
> 
> I would make something span like, that in case of scalar record would have `size = 1`.
> In this way the `getUsedMemory` would be `sizeof(span<T>) + sizeof(T)*m_datum.size()`
> 
> We can't use the the `std::span` but we can use it something like this (https://github.com/boostorg/beast/blob/develop/include/boost/beast/core/span.hpp) or implement our own `yarp::telemetry::Span`
> 
> What do you think about it?
> 
> cc @traversaro @AlexAntn @S-Dafarra @drdanz

``Span`` may be an option. Note that there are limitations. For example, it does not support boolean, and in general it works only with array-like containers.

Regarding the implementation of ``Span``, it is often easier to just copy it around, being a single header file. See for example:
- https://github.com/robotology/idyntree/blob/master/src/core/include/iDynTree/Core/Span.h
- https://github.com/dic-iit/matio-cpp/blob/master/include/matioCpp/Span.h

Those are both inherited from https://github.com/Microsoft/GSL/blob/master/include/gsl/span

The nice thing is that, thanks to constructors like https://github.com/dic-iit/matio-cpp/blob/master/include/matioCpp/Span.h#L397-L413, they are basically interchangeable (you can implicitly pass from one to the other).

traversaro, 2021-01-08:
I don't think it make sense to have Span to be the type used by Record. If I remember correctly, the Record is meant to storage type used by the circular buffer, and as such it should be something that owns the memory. `std::span` and all its vendored version are quite nice and I think it could make sense to use them in the user-facing API, but in the process of inserting the data in the circular buffer, the data pointed by the span provided by the user should be copied to some concrete data structure that owns its memory, for example std::vector .

drdanz, 2021-01-11:
If you want to use span, you might also consider to import this: https://github.com/tcbrindle/span (Boost Software License, Version 1.0).
But in this case I agree with @traversaro

PR #31 (2021-01-18): Add BufferManager API
It fixes #29 
It fixes #23
 
This PR introduces these changes:

- Use `double` as timestamp in `Record<T>` (see https://github.com/robotology-playground/yarp-telemetry/issues/28#issuecomment-758003434)
- Use `std::vector<T>` as type for `m_datum` of `Record<T>`(see https://github.com/robotology-playground/yarp-telemetry/issues/23#issuecomment-756782927).
- Implements https://github.com/robotology-playground/yarp-telemetry/pull/27#issuecomment-758001061
- Add `BufferManager` class

This is a template class that contains a `std::map<string, telemetry::Buffer<T>>` that conta [... cut]
closed 2021-01-18 by Nicogene