robotology/robometry #26: Select a real world codebase for a first proof of concept

traversaro, 2021-01-05:


traversaro, 2021-01-05:
We could look into something used in real world, even if simple. We can take something from https://github.com/robotology/icub-basic-demos or from the vvv tutorial. Ideally we should not do a example from no where, as we risk of structuring the code around our available APIs, rather then understanding how our APIs should be done to be convenient to use on real world  code.

traversaro, 2021-01-05:
A relatively simple example for sure used in real world is https://github.com/robotology/icub-tests/blob/v1.18.0/src/positionDirect/PositionDirect.cpp#L243, but perhaps the RTF machine may be complicate its use. A more didactic example is https://github.com/vvv-school/tutorial_joint-interface/blob/4d71c7846b61ad321b0f53f8c5a09b6880a25d1c/src/positiondirect.cpp#L171.

traversaro, 2021-01-22:
Another interesting possible use is https://github.com/robotology/gazebo-yarp-plugins/pull/529 , in which a simple test for the `gazebo_yarp_controlboard` is added, and in which the telemetry could be used to gather data on measured vs desired position and velocity, in place of simple prints https://github.com/robotology/gazebo-yarp-plugins/pull/529/files#diff-5a4a105a9b072d936198482927cfd93bd43bee3159bd74640f4c1d2acf77ef48R115 .

Nicogene, 2021-01-25:
Today I tried including the `yarp::telemetry::BufferManager` inside https://github.com/vvv-school/tutorial_joint-interface, in particular in `tutorial_joint-interface-position`.

Here are the modifications:

```diff
--- a/src/position.cpp
+++ b/src/position.cpp
@@ -12,6 +12,7 @@
 #include <yarp/os/RFModule.h>
 #include <yarp/os/RpcServer.h>
 #include <yarp/os/Time.h>
+#include <yarp/telemetry/BufferManager.h>

 #include <yarp/dev/Drivers.h>
 #include <yarp/dev/PolyDriver.h>
@@ -20,6 +21,8 @@
 using namespace yarp::os;
 using namespace yarp::dev;

+yarp::telemetry::BufferManager<double> bm("tutorial_pos_ctrl.mat",
+                                          { {"pos",{1,1}} }, 100, true);

 class CtrlModule: public RFModule
 {
@@ -69,6 +72,8 @@ protected:
         double t0=Time::now();
         while (!done&&(Time::now()-t0<10.0))
         {
+            ienc->getEncoder(joint,&enc);
+            bm.push_back({enc}, "pos");
             yInfo()<<"Waiting...";
             Time::delay(0.1);   // release the quantum to avoid starving resources
             ipos->checkMotionDone(&done);
```

## Result
Here is the resulting struct:

```
>> tutorial_pos_ctrl.pos

ans = 

  struct with fields:

          data: [1×1×67 double]
    dimensions: [67 1 1]
          name: 'pos'
    timestamps: [1×67 double]
```

## Issues
- Using it in an external code I found out quite cumbersome to define everything in the constructor (see #36 )
- Moreover, I had issues with not having enough samples dumped(100 in this example), see https://github.com/robotology-playground/yarp-telemetry/blob/d4d586d85487d87c644a44824b993d00a8607f4f/src/libYARP_telemetry/src/yarp/telemetry/BufferManager.h#L66-L70

traversaro, 2021-01-25:
> Moreover, I had issues with not having enough samples dumped(100 in this example), see

Tracked in https://github.com/robotology-playground/yarp-telemetry/issues/46 .

traversaro, 2021-01-25:
@Nicogene can you paste the link to the fork/branch of the modified version of the example, and the we can close the issue?

Nicogene, 2021-01-25:
Here is the branch where I pushed these changes https://github.com/Nicogene/tutorial_joint-interface/tree/yarp_telemetry

Closing.

closed 2021-01-25 by Nicogene