robotology/robometry #28: Decide what to use as timestamp

Nicogene, 2021-01-08:
For now `yarp::telemetry::Record` uses `yarp::os::Stamp` that has some pro and cons

## Pro

- Easily writable to the yarp network
- Provide both reference count and timestamp

## Cons

- If we don't need the reference count,  maybe a `double` is sufficient
- We are stick to the dependency of a yarp object for expressing something that is already available in the standard

On the other hand, I have to say that the std::chrono is not super user friendly, with its cumbersome types.

traversaro, 2021-01-08:
I think we can probably reach the MVP with what we have (for now yarp::os::Stamp), and then see the limitations of one approach vs the other in the MVP.

drdanz, 2021-01-11:
> On the other hand, I have to say that the `std::chrono` is not super user friendly, with its cumbersome types.

😱 I think you are wrong here, since the comparison is with a `double` which does not make differences between time point and duration. Anyway, not the actual point of the discussion 😆 

> Provide both reference count and timestamp

Just *count*, not *reference count*...

I just wanted to mention that I believe that using the `Stamp` might not be the best option, since if you have 3 records in a buffer, and each record has its `Stamp`, you will still have to keep the counter externally, because the `update()` of each `Stamp` will be called only 1/3 of the times. I'd rather go with a `double` and, if you need counting, add a counter to the buffer class

traversaro, 2021-01-11:
For now we will use `double`.

closed 2021-01-11 by traversaro