robotology/robometry #37: Collect features that are missing for DIC use of this library

traversaro, 2021-01-19:


S-Dafarra, 2021-01-25:
I took the time to give another look at the code. I think it is a good start. Right now, it allows storing a fixed set of ``std::vector`` and save it to a file. The input ``std::vector`` can also be treated as a matrix.

I think this set of functionalities might be good for some new pieces of software, where the cost of including a new dependency is lower than working out a custom solution. In addition, if such a piece of software is in active development, it is not a big deal to update it as soon as the API of the telemetry changes.

IMHO, the (meta) functionalities that may drive to a heavier use of this library might be:
- "Dependability"
    - A virtually unlimited buffer dimension. An experiment may last from 30sec to a couple of hours (as it happens during fairs or stress tests). If we are not sure this tool may back us for the entirety of the experiment, we will most likely use other tools.
    - The data is saved even in case the application crashes. This might be a killer feature, especially for the most complex code that might be crashing in specific conditions that are very hard to spot
- "Flexibility"
    - Right now the interface requires a reference to a ``std::vector`` as input. This is not the most flexible data type when it comes to vectors. In our applications, we use ``std::vectors`` just for a narrow subset of data. For all the other cases, we would need an intermediate copy, which is not so convenient. In order to avoid dynamic memory allocation in time-critical methods, we would need to use a ``std::vector`` as a class parameter for each variable we want to store, which is not very practical.  
    - Right now, the configuration of the buffer can be done only in the constructor. Most of the time, we know what variable we want to log only after the configuration phase at runtime.
- "Scalability"
    - In one of our most recent demo, we need to run tens of applications spread on 4-5 different PCs across 2 networks. If every single application generates a file, dealing with all of them might become tricky, especially if it is necessary to observe behaviors that arise from the interaction of two or more applications. Some automatic mechanism that can gather and merge these output files might be a killer feature to start spreading this tool around.

traversaro, 2021-01-25:
fyi @maggia80 @DanielePucci

traversaro, 2021-01-25:
Thanks @S-Dafarra ! Could you assign a rough priority/importance level to the five different aspects that you listed? I don't want to give the impression that this will directly translate in the exact order of implementation (as the complexity also plays a role), but for sure it would help planning.

S-Dafarra, 2021-01-25:
> Thanks @S-Dafarra ! Could you assign a rough priority/importance level to the five different aspects that you listed? I don't want to give the impression that this will directly translate in the [...]

From the user perspective, time is not the key at the moment as there is always the "quick and dirty custom implementation" option. The point then is how much we can make the telemetry option more "attractive" than the "quick and dirty custom implementation" one. In this sense, the "killer features" (the second and last point) might be the most important, since they are the most complicated to be replicated in the "quick and dirty custom implementation". In a sense, the nuisance of writing workarounds for the other points (in case they are not solved) would be paid off by the advantages of having those two points in place.

traversaro, 2021-01-25:
@AlexAntn @Nicogene could you open an issue for each one of this points? When this is done, I think we can close the issue.

Nicogene, 2021-01-26:
All the @S-Dafarra points are now tracked in separate issues.
Closing.

closed 2021-01-26 by Nicogene