robotology/robometry #39: Investigation file-based configuration of the buffer manager

traversaro, 2021-01-19:


Nicogene, 2021-02-05:
I found a very nice c++ library for parsing JSON files https://github.com/nlohmann/json

In particular, it can be imported also as a single header(see https://github.com/nlohmann/json/tree/develop/single_include/nlohmann) and in particular [this feature](https://github.com/nlohmann/json#arbitrary-types-conversions) is awesome.

We may implement the `from_json` and `to_json` in our `BufferConfig` struct!

cc @traversaro @AlexAntn @S-Dafarra

traversaro, 2021-02-05:
Good idea! I would however implement it in separate function/headers, so that we can easily add more conversions in the future, for example from/to yarp::os::Property as well, and permit to users to provide their own conversions, as adding more functions is easy, while adding methods to an existing class is impossible outside of the project.

cc @GiulioRomualdi

traversaro, 2021-02-05:
Regarding the use of single header, we can also support using like that (especially if we want to support 18.04 that has an old version: https://repology.org/project/nlohmann-json/versions), but I would have at least an option to find it externally via `find_package(nlohmann_json)`.

Nicogene, 2021-02-05:
In #65 we implemented the configuration from json, and that library is really easy to use!
For what we understood the default behaviour is that all the fields have to be specified in the .json file, no one is taken as optional

We have to investigate how to handle the case in which the user forgot a parameter

S-Dafarra, 2021-02-05:
It would be nice to have keywords like ``all`` to store all the parameters, or ``not`` to put some specific parameter in the blacklist.

closed 2021-02-09 by Nicogene

barbalberto, 2021-02-09:
I like that library as well, has a modern approach to C++.
The header only version is handy, except when you start using all over your
coffee it slows down compilation time.

Also uses a lot of exceptions whenever anything is not perfect, which
personally I don't like so much.

Also, the boost::ptree is a good alternative to parse json file

Nicogene, 2021-02-09:
> Also uses a lot of exceptions whenever anything is not perfect, which
personally I don't like so much.

Me neither, when the file misses some attribute is not clear what is missing or what it is wrong

commit 70a34b97 (2021-02-08): BufferManager: add conversion from and to json for the BufferConfig struct

We have to move these in different header
It fixes #39
commit b0624cde (2021-02-08): BufferManager: add conversion from and to json for the BufferConfig struct

We have to move these in different header
It fixes #39
PR #65 (2021-02-09): BufferManager: add mutex, configuration from file, empty constructor workflow et al
- The `ChannelInfo` has been changed from a struct of two elements to a
`std::pair`, in order to be easily handled with the JSON file conf.
- All the flags, options of the BufferManager have been put in the BufferConfig
- The empty constructor is allowed now, for this we added all the functions
needed for configuring a default BufferManager(It fixes #60)

- The mutex has been added in `saveToFile`(partially address #9)
- Now the periodicSave, saves only the variables that go above the
data_treshold specified
-  [... cut]