robotology/robometry #43: Make more clear example of BufferManager

Nicogene, 2021-01-22:
We should extend the examples about the `BufferManager` and make them more clear(e.g. less modern c++ black magic)

traversaro, 2021-01-22:
A step in both solving this issue and solving https://github.com/robotology-playground/yarp-telemetry/issues/39 would be to move the configuration of the BufferManager from being done in the constructor, to be done via appropriate configuration methods.

pattacini, 2021-01-22:
It's not black magic 😏 
It's more about making the example clearer.

Replacing
```c++
yarp::telemetry::BufferManager<double> bm_v({ {"one",{4,1}}, {"two",{4,1}} }, 3);
```
with
```c++
// configure telemetry to log 3 samples of 2 variables of vector type whose size is 4 double each
// the channels are called "one" and "two"
size_t n_samples = 3;
std::vector<BufferInfo> listOfVars{ {"one", {4,1}}, {"two", {4,1}} };
yarp::telemetry::BufferManager<double> bm_v(listOfVars, n_samples);
```
enhances readability.

In my opinion, it's important to introduce the concept of _channel_, which is not the variable per se to be logged but rather refers to the infrastructure employed to transfer the information.

commit 3197072a (2021-01-27): examples: use new BufferManager methods

It fixes #43
closed 2021-01-28 by Nicogene