robotology/robometry #44: Linking problems for BufferManager

Nicogene, 2021-01-25:
I tried to include `YARP_telemetry` in an external project and I get this linking error

```
/usr/bin/ld: CMakeFiles/tutorial_joint-interface-position.dir/src/position.cpp.o: in function `matioCpp::Vector<char>::checkCompatibility(matvar_t const*, matioCpp::VariableType, matioCpp::ValueType) const':
position.cpp:(.text._ZNK8matioCpp6VectorIcE18checkCompatibilityEPK8matvar_tNS_12VariableTypeENS_9ValueTypeE[_ZNK8matioCpp6VectorIcE18checkCompatibilityEPK8matvar_tNS_12VariableTypeENS_9ValueTypeE]+0x84): undefined reference to `matioCpp::get_types_names_from_matvart(matvar_t const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
/usr/bin/ld: CMakeFiles/tutorial_joint-interface-position.dir/src/position.cpp.o: in function `matioCpp::MultiDimensionalArray<double>::checkCompatibility(matvar_t const*, matioCpp::VariableType, matioCpp::ValueType) const':
position.cpp:(.text._ZNK8matioCpp21MultiDimensionalArrayIdE18checkCompatibilityEPK8matvar_tNS_12VariableTypeENS_9ValueTypeE[_ZNK8matioCpp21MultiDimensionalArrayIdE18checkCompatibilityEPK8matvar_tNS_12VariableTypeENS_9ValueTypeE]+0x78): undefined reference to `matioCpp::get_types_names_from_matvart(matvar_t const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
/usr/bin/ld: CMakeFiles/tutorial_joint-interface-position.dir/src/position.cpp.o: in function `matioCpp::Vector<int>::checkCompatibility(matvar_t const*, matioCpp::VariableType, matioCpp::ValueType) const':
position.cpp:(.text._ZNK8matioCpp6VectorIiE18checkCompatibilityEPK8matvar_tNS_12VariableTypeENS_9ValueTypeE[_ZNK8matioCpp6VectorIiE18checkCompatibilityEPK8matvar_tNS_12VariableTypeENS_9ValueTypeE]+0x78): undefined reference to `matioCpp::get_types_names_from_matvart(matvar_t const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
/usr/bin/ld: CMakeFiles/tutorial_joint-interface-position.dir/src/position.cpp.o: in function `matioCpp::Vector<double>::checkCompatibility(matvar_t const*, matioCpp::VariableType, matioCpp::ValueType) const':
position.cpp:(.text._ZNK8matioCpp6VectorIdE18checkCompatibilityEPK8matvar_tNS_12VariableTypeENS_9ValueTypeE[_ZNK8matioCpp6VectorIdE18checkCompatibilityEPK8matvar_tNS_12VariableTypeENS_9ValueTypeE]+0x78): undefined reference to `matioCpp::get_types_names_from_matvart(matvar_t const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
/usr/bin/ld: CMakeFiles/tutorial_joint-interface-position.dir/src/position.cpp.o: in function `std::vector<matioCpp::Variable, std::allocator<matioCpp::Variable> >::~vector()':
position.cpp:(.text._ZNSt6vectorIN8matioCpp8VariableESaIS1_EED2Ev[_ZNSt6vectorIN8matioCpp8VariableESaIS1_EED5Ev]+0x28): undefined reference to `matioCpp::Variable::~Variable()'
/usr/bin/ld: CMakeFiles/tutorial_joint-interface-position.dir/src/position.cpp.o: in function `matioCpp::Vector<double>::operator=(matioCpp::Span<double, -1l> const&)':
position.cpp:(.text._ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE[_ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE]+0x29): undefined reference to `matioCpp::Variable::dimensions() const'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE[_ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE]+0x34): undefined reference to `matioCpp::Variable::dimensions() const'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE[_ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE]+0x6a): undefined reference to `matioCpp::Variable::name[abi:cxx11]() const'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE[_ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE]+0xa6): undefined reference to `matioCpp::Variable::initializeVariable(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, matioCpp::VariableType const&, matioCpp::ValueType const&, matioCpp::Span<unsigned long const, -1l>, void*)'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE[_ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE]+0xec): undefined reference to `matioCpp::Variable::dimensions() const'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE[_ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE]+0xf7): undefined reference to `matioCpp::Variable::dimensions() const'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE[_ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE]+0x113): undefined reference to `matioCpp::Variable::toMatio()'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE[_ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE]+0x134): undefined reference to `matioCpp::Variable::dimensions() const'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE[_ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE]+0x13f): undefined reference to `matioCpp::Variable::dimensions() const'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE[_ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE]+0x15c): undefined reference to `matioCpp::Variable::dimensions() const'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE[_ZN8matioCpp6VectorIdEaSERKNS_4SpanIdLln1EEE]+0x167): undefined reference to `matioCpp::Variable::dimensions() const'
/usr/bin/ld: CMakeFiles/tutorial_joint-interface-position.dir/src/position.cpp.o: in function `matioCpp::Vector<int>::operator=(matioCpp::Span<int, -1l> const&)':
position.cpp:(.text._ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE[_ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE]+0x29): undefined reference to `matioCpp::Variable::dimensions() const'
/usr/bin/ld: CMakeFiles/tutorial_joint-interface-position.dir/src/position.cpp.o:position.cpp:(.text._ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE[_ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE]+0x34): more undefined references to `matioCpp::Variable::dimensions() const' follow
/usr/bin/ld: CMakeFiles/tutorial_joint-interface-position.dir/src/position.cpp.o: in function `matioCpp::Vector<int>::operator=(matioCpp::Span<int, -1l> const&)':
position.cpp:(.text._ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE[_ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE]+0x6a): undefined reference to `matioCpp::Variable::name[abi:cxx11]() const'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE[_ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE]+0xa6): undefined reference to `matioCpp::Variable::initializeVariable(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, matioCpp::VariableType const&, matioCpp::ValueType const&, matioCpp::Span<unsigned long const, -1l>, void*)'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE[_ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE]+0xec): undefined reference to `matioCpp::Variable::dimensions() const'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE[_ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE]+0xf7): undefined reference to `matioCpp::Variable::dimensions() const'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE[_ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE]+0x113): undefined reference to `matioCpp::Variable::toMatio()'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE[_ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE]+0x134): undefined reference to `matioCpp::Variable::dimensions() const'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE[_ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE]+0x13f): undefined reference to `matioCpp::Variable::dimensions() const'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE[_ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE]+0x15c): undefined reference to `matioCpp::Variable::dimensions() const'
/usr/bin/ld: position.cpp:(.text._ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE[_ZN8matioCpp6VectorIiEaSERKNS_4SpanIiLln1EEE]+0x167): undefined reference to `matioCpp::Variable::dimensions() const'
[... cut ...]
/usr/bin/ld: CMakeFiles/tutorial_joint-interface-position.dir/src/position.cpp.o:(.data.rel.ro._ZTIN8matioCpp21MultiDimensionalArrayIdEE[_ZTIN8matioCpp21MultiDimensionalArrayIdEE]+0x10): undefined reference to `typeinfo for matioCpp::Variable'
/usr/bin/ld: CMakeFiles/tutorial_joint-interface-position.dir/src/position.cpp.o:(.data.rel.ro._ZTIN8matioCpp6VectorIcEE[_ZTIN8matioCpp6VectorIcEE]+0x10): undefined reference to `typeinfo for matioCpp::Variable'
collect2: error: ld returned 1 exit status
```

Nicogene, 2021-01-25:
This can be fixed by linking the external executable to `matioCpp::matioCpp`
This is why we ought to address #34

closed 2021-01-25 by Nicogene

traversaro, 2021-01-25:
> 
> 
> This can be fixed by linking the external executable to `matioCpp::matioCpp`
> This is why we ought to address #34

Even before addressing #34 , we should add matioCpp in YARP_telemetry_PUBLIC_DEPS  https://github.com/robotology-playground/yarp-telemetry/blob/4fcbe4066225f02688292dcd002440ec3aed840a/src/libYARP_telemetry/src/CMakeLists.txt#L57