robotology/robometry #49: Investigate about a more flexible data type for storing data

Nicogene, 2021-01-25:
From #37:
> Right now the interface requires a reference to a std::vector as input. This is not the most flexible data type when it comes to vectors. In our applications, we use std::vectors just for a narrow subset of data. For all the other cases, we would need an intermediate copy, which is not so convenient. In order to avoid dynamic memory allocation in time-critical methods, we would need to use a std::vector as a class parameter for each variable we want to store, which is not very practical.

In particular, we have to consider that is related to what is under the hood of `yaro::telemetry::Record`
A similar discussion has been done in #23.

GiulioRomualdi, 2021-02-04:
Hi all, sorry to jump in. You may think to use [`iDynTree::Span`](https://github.com/robotology/idyntree/blob/master/src/core/include/iDynTree/Core/Span.h) (for vectors) and  [`iDynTree::MatrixView`](https://github.com/robotology/idyntree/blob/master/src/core/include/iDynTree/Core/MatrixView.h) for matrices. 

I think that you can just copy-paste the two files in the repo since they are only two headers. 

cc @traversaro

Nicogene, 2021-03-02:
In the first application of the library(#84) the `std::vector` under the hood of the `Record` proved quite handy, since the yarp interfaces for retrieving the encoders, speed etc are `std::vector`-compatible.

S-Dafarra, 2021-12-02:
Let me revive this issue. I think that moving to ``Span`` as @GiulioRomualdi suggested could be a good way to avoid temporary copies when the data to store is not an ``std::vector``

GiulioRomualdi, 2021-12-08:
I started to convert the `std::vector` in `matioCpp::Span` however, I had to add `matioCpp::make_span()` in all the examples. For instance
```cpp
cb_v.push_back(Record(yarp::os::Time::now(), vector<double>{0.9, 1.0, 1.1}));
```
becomes 
```cpp
cb_v.push_back(Record(yarp::os::Time::now(), matioCpp::make_span(vector<double>{0.9, 1.0, 1.1})));
```
In theory the conversion between `std::vector<>` into `std::span` should be implicit given the Constructor (7)  [here](https://en.cppreference.com/w/cpp/container/span/span) and [here](https://stackoverflow.com/questions/64900594/how-can-stdvector-be-converted-to-stdspan). However, the `span` implementation stored in MatioCpp does not have that constructor. 

What do you suggest? At this point, I see 3 possible solutions:
1. use `make_span` where required
2. Implement the missing constructor in `matioCpp::Span` (I should understand how)
3. Ask for c++20. I checked that `std::span` is available in `gcc-10` that is available in ubuntu 20.04 but it is not installed by `build-essential`
   [image]

S-Dafarra, 2021-12-09:
Hold on, in the code you wrote above, you are creating a span from a temporary object.

GiulioRomualdi, 2021-12-09:
Yes but it shouldn't be a problem since I'm creating a `span<const int>`

S-Dafarra, 2021-12-09:
> Yes but it shouldn't be a problem since I'm creating a `span<const int>` and the lifespan of the vector ends at the end of the function.
> 
> I cam try with the initialized list only
> 
> See here https://newbedev.com/why-can-t-i-construct-a-gsl-span-with-a-brace-enclosed-initializer-list

From the very same article you linked :wink: :

> Span is non owning. Does not own storage. It is a replacement for pointer arithmetic, not a storage class.

>You cannot initialize a span with an initializer list, because there is nowhere to put the data.

``vector<double>{0.9, 1.0, 1.1})`` gets deallocated as soon as ``make_span`` returns, hence the ``double*`` stored inside ``span`` returned by ``make_span`` would be pointing to deallocated memory.

S-Dafarra, 2021-12-09:
In any case, those constructors you are mentioning should be already there: https://github.com/ami-iit/matio-cpp/blob/d8a450ea7c3289ca78ea5111aea0f451a8b19b74/include/matioCpp/Span.h#L397-L413

GiulioRomualdi, 2021-12-09:
So why is it not implicit constructible? 🤔

GiulioRomualdi, 2021-12-09:
> `vector<double>{0.9, 1.0, 1.1})` gets deallocated as soon as `make_span` returns, hence the `double*` stored inside `span` returned by `make_span` would be pointing to deallocated memory.

So we should avoid to use `make_span`

S-Dafarra, 2021-12-09:
> So why is it not implicit constructible? thinking

Can you share the code and the error you get?

> So we should avoid to use `make_span`

No, we should not use temporary objects. Simply mode the definition of the vector a line above (that is a more common use case by the way)

GiulioRomualdi, 2021-12-09:
> No, we should not use temporary objects. Simply mode the definition of the vector a line above (that is a more common use case by the way)

By the way, what I was looking for is
```cpp
void foo(span<const int> s)
{
};

foo(vector<int>{1, 2, 3});
```
and this should be permitted, am I missing something?

S-Dafarra, 2021-12-09:
> > No, we should not use temporary objects. Simply mode the definition of the vector a line above (that is a more common use case by the way)
> 
> By the way, what I was looking for is
> 
> ```c++
> void foo(span<const int> s)
> {
> };
> 
> 
> foo(vector<int>{1, 2, 3});
> ```
> 
> and this should be permitted, am I missing something?

You are still creating a temporary object. This should be permitted though
```cpp
void foo(span<const int> s)
{
};

std::vector<int> v = {1, 2, 3};
foo(v);
```

GiulioRomualdi, 2021-12-09:
I tried to compile this code on my PC 
```cpp
#include <iostream>
#include <span>
#include <vector>

void foo(std::span<const int> span)
{
    for (const auto& s : span)
    {
        std::cout << s << std::endl;
    }
};

int main()
{
    foo(std::vector<int>{1, 2, 3, 4});

    return 0;
}
```
with the following command `g++-10 -g -std=c++20 test.cpp` 
everything works fine and valgrind says 
```
==23354== Memcheck, a memory error detector
==23354== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==23354== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==23354== Command: ./a.out
==23354== 
=====================================================================================================
1
2
3
4
=====================================================================================================
==23354== 
==23354== HEAP SUMMARY:
==23354==     in use at exit: 0 bytes in 0 blocks
==23354==   total heap usage: 3 allocs, 3 frees, 76,816 bytes allocated
==23354== 
==23354== All heap blocks were freed -- no leaks are possible
==23354== 
==23354== For lists of detected and suppressed errors, rerun with: -s
==23354== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

I think it is working because the memory is kept allocated since `foo()` returns.

It is the same as what happens when you pass a temporary object to a class that accepts `const &`
```cpp
#include <iostream>
#include <vector>

void foo(const std::vector<int>& vector)
{
    for (const auto& v : vector)
    {
        std::cout << v << std::endl;
    }
};

int main()
{
    foo(std::vector<int>{1, 2, 3, 4});

    return 0;
}
```

S-Dafarra, 2021-12-09:
Note, you are leaving to the compiler optimizations the choice of whether that code will crash or not. As per https://en.cppreference.com/w/cpp/language/lifetime

> All temporary objects are destroyed as the last step in evaluating the full-expression that (lexically) contains the point where they were created, and if multiple temporary objects were created, they are destroyed in the order opposite to the order of creation. This is true even if that evaluation ends in throwing an exception.
> 
> There are two exceptions from that:
> 
> The lifetime of a temporary object may be extended by binding to a const lvalue reference or to an rvalue reference (since C++11), see reference initialization for details.
> The lifetime of a temporary object created when evaluating the default arguments of a default or copy constructor used to initialize an element of an array ends before the next element of the array begins initialization.

Note **may**, it is not a constraint. It is most probably an undefined behavior.

PR #144 (2021-12-14): Introduce the `matioCpp::Span` in the interface of yarp-telemetry
This closes #49 

this is the first step towards https://github.com/ami-iit/bipedal-locomotion-framework/issues/189

cc @traversaro @S-Dafarra @Nicogene
closed 2021-12-14 by Nicogene