robotology/robometry #51: Investigate the implementation of a daemon/collector for all the .mat files generated by the telemetry

AlexAntn, 2021-01-26:


AlexAntn, 2021-06-11:
There are multiple ways we can handle this. Below is a list of the possible structures that could be used, their shortcomings and advantages:

## Daemons

A daemon will interact with the different instances of telemetry running in the machine cluster. It would probably require that we start the telemetry clients with a daemon IP extra configuration variable, or something similar.

We have two options here:

- A **continuous** daemon, where instead (or in parallel) of writing the logs to files, it sends the data to the daemon to be logged in a centralized way;
- A **after experiment** daemon, where the logs are sent only when the telemetry client is closed, **OR** when it tells the daemon where to collect the logs from (IP + path).

In either of these cases, we have the option of a more informative daemon, or merely an informed collector. In the case of the more informative daemon, we could have a GUI displaying the collected data (the degree of detail/finesse of the GUI is not relevant to define the structure for now)

**Advantages:**
- We can merge the logfiles into a single one
- We can collect the logs periodically (avoiding the space issues of machines with limited space)
- Allows the users to visualize the data that has been collected (if we decide to integrate a GUI).
- It doesn't require a standard path for the logs (The API/client would provide the daemon with it)

**Shortcomings:**
- It requires us to change the API/client to interact with the daemon
- It requires the user to run a module (before the API/clients are launched)
- It requires the user to specify the IP of the daemon machine for each telemetry instance (in config file or some other way)

## Collectors

A collector will not interact with the clients at all. It could be executed at any point of an experiment, provided with a list of the machines/IPs to check for logs, and collect all logs it finds into a single machine.

For the collector to work, we would have to assume a standard path for the telemetry logs to be written to. Alternatively and making it more complex, we could specify the IP and path for each machine. 

The collector module could, like the daemon option, have a GUI that would allow the users to visualize the data.

**Advantages:**
- It doesn't require us to change the API/client (the collector takes care of everything)
- We can merge all logfiles into a single one
- We can collect the logs periodically
- Allows the users to visualize the data that has been collected (if we decide to integrate a GUI).

**Shortcomings:**
- It requires either a fixed path to store the logs in each machine, or to specify for each individual machine not only its IP but also the path to take the logs from;
- It requires the user to run a module (independent of client)

## Default Machine

Another way to handle the collection of logs is to specify a standard machine and path to write logs to, for each client. This could be set in the configuration file for the client. Once an experiment is completed, the client would forward the logs to that machine/path itself. 

This means no extra modules need to be run (good), but the client would have to be changed (bad), and you would have no module to visualize the data (bad)

**Advantages:**
- User doesn't have to launch an extra module
- No fixed paths needed

**Shortcomings:**
- Harder to merge the logs (they are coming from many different places
- No GUI option possible
- We have to specify for each client which IP and path to user (the one of the "central" machine)

____________________________________________________________________________

My personal opinion is that the **collector** option is the one that makes more sense, since it doesn't break the existing API/client, should still be able to collect all the logs, and is able to display a GUI with the data should we opt to go down that route.

I would personally avoid changing the API as much as possible, since it is working well in its current instance.

drdanz, 2021-06-11:
My 2 cents... you should collect data and then generate files containing the data you need, not files.