robotology/robometry #54: Add unit test in the CI

Nicogene, 2021-01-28:
We are in a stage where the API change often so it is difficult to implement and maintain unit tests, but at some point, we should introduce unit tests in the CI.

My proposal is to use [Catch2](https://github.com/catchorg/Catch2)

GiulioRomualdi, 2021-02-01:
An example of usage:

- `Cmake` https://github.com/dic-iit/bipedal-locomotion-framework/blob/master/cmake/Catch2Main.cpp.in and  https://github.com/dic-iit/bipedal-locomotion-framework/blob/master/cmake/AddBipedalLocomotionUnitTest.cmake
- `cpp`: https://github.com/dic-iit/bipedal-locomotion-framework/blob/master/src/ParametersHandler/tests/ParametersHandlerTest.cpp

traversaro, 2021-02-03:
To avoid regression such as https://github.com/robotology-playground/yarp-telemetry/issues/59, a low effort modification is also to run examples as tests, just to ensure that they run fine and eventually run them under valgrind to check if they don't memory errors.

PR #78 (2021-02-19): Add first draft of unit tests
It fixes #54 

This PR add the first test case `BufferManagerTest` that tests all the functionalities we added so far to the `BufferMananger` so far.

There are still open points e.g:

- I had to add delays in order to avoid segfaults due to the destruction of `BufferManager` object too soon(#9 is not completely fixed, we have to investigate about it cc @AlexAntn )
~- Locally I had issues with `nlohmann_json`, enabling the tests enabled also the json unit tests, I want to see if this problem is happening also in the CI.~
This seems to not happen if using a clean w [... cut]
closed 2021-02-19 by Nicogene