robotology/robometry #66: json parser: use fetch content

Nicogene, 2021-02-05:
In #65 we added the json.hpp in the repo but it would be better to use cmake FetchContent

traversaro, 2021-02-05:
A typical boilerplate for this is:
~~~
option(YARP_TELEMETRY_USES_SYSTEM_nlohmann_json OFF)
if(YARP_TELEMETRY_USES_SYSTEM_nlohmann_json)
  find_package(nlohmann_json REQUIRED)
else()
  include(FetchContent)

  FetchContent_Declare(json
    GIT_REPOSITORY https://github.com/nlohmann/json.git
    GIT_TAG v3.7.3)

  FetchContent_GetProperties(json)
  if(NOT json_POPULATED)
    FetchContent_Populate(json)
    add_subdirectory(${json_SOURCE_DIR} ${json_BINARY_DIR} EXCLUDE_FROM_ALL)
  endif()
endif()
~~~ 

and then in the CMake code you just use `nlohmann_json::nlohmann_json` , see https://github.com/nlohmann/json#embedded-fetchcontent .

Nicogene, 2021-02-05:
I completely missed this https://github.com/nlohmann/json#embedded-fetchcontent !

commit 6f9d83fc (2021-02-08): YARP_telemety: add nlohmann_json parser dependency

It fixes #66
commit 486b96d3 (2021-02-08): YARP_telemety: add nlohmann_json parser dependency

It fixes #66
PR #65 (2021-02-09): BufferManager: add mutex, configuration from file, empty constructor workflow et al
- The `ChannelInfo` has been changed from a struct of two elements to a
`std::pair`, in order to be easily handled with the JSON file conf.
- All the flags, options of the BufferManager have been put in the BufferConfig
- The empty constructor is allowed now, for this we added all the functions
needed for configuring a default BufferManager(It fixes #60)

- The mutex has been added in `saveToFile`(partially address #9)
- Now the periodicSave, saves only the variables that go above the
data_treshold specified
-  [... cut]
closed 2021-02-09 by Nicogene