robotology/robometry #76: Investigate about the possibility to make the BufferMananger manage different data types

Nicogene, 2021-02-15:
At this moment the `BufferManager` is a template class, then it can handle only one type of data.
It would be nice if we can implement only template functions(maybe moving them outside the class) in order to make it possible to handle different data types with just one `BufferManager`(maybe it would become a singleton?)

Nicogene, 2021-02-15:
This goes in #34 direction

S-Dafarra, 2022-01-10:
During a meeting we had with @traversaro @GiulioRomualdi @AlexAntn and @Nicogene, we thought that it might be indeed interesting to have this fixed. The main reasons are:
- the possibility to store and synchronize different types of data:
   - double
   - int
   - strings
- remove the template from the BufferManager class.

One possibility would be to use https://en.cppreference.com/w/cpp/utility/any or https://en.cppreference.com/w/cpp/utility/variant

S-Dafarra, 2022-01-10:
I just figured that rather than using variant/any up to ``Record``

we can edit this line
https://github.com/robotology/yarp-telemetry/blob/0646ef0d6c099b5a3c6291a0f7d535c46a7462e8/src/libYARP_telemetry/src/yarp/telemetry/experimental/BufferManager.h#L594
into 
``
std::shared_ptr<TreeNode<std::any>> m_tree;
``
if possible. Then, ``std::any`` could be ``BufferInfo<double>``, ``BufferInfo<int>``, ```BufferInfo<sd::string>``, ...

In this way, the modification remain limited to the ``BufferManager`` class only.

Nicogene, 2022-01-11:
I would prefer using `variant` instead of `any`, this post resume why it is better and safer https://stackoverflow.com/questions/56303939/c-stdvariant-vs-stdany

S-Dafarra, 2022-01-11:
The types supported by ``matioCpp`` are those listed in https://github.com/ami-iit/matio-cpp/blob/d8a450ea7c3289ca78ea5111aea0f451a8b19b74/include/matioCpp/ConversionUtilities.h#L50-L62. To this list, we can add ``std::string`` that is automatically turned into a vector of chars (see https://github.com/ami-iit/matio-cpp/blob/d8a450ea7c3289ca78ea5111aea0f451a8b19b74/include/matioCpp/ForwardDeclarations.h#L192)

S-Dafarra, 2022-01-18:
In the direction of being compatible with different data types, a possibility could be also to store videos. I suppose the type of the channel, in this case, could be some sort of image (or even a char array). Related issue #155

S-Dafarra, 2022-03-30:
I am currently working on this in https://github.com/S-Dafarra/yarp-telemetry/tree/multiple_types

In the end, I opted for ``std::any``. The main reason is that with https://github.com/ami-iit/matio-cpp/pull/55 we can convert generic structs to ``matioCpp::Struct``s. Hence, I cannot know a prior all the types that we could store.

S-Dafarra, 2022-04-28:
Corresponding PR: https://github.com/robotology/yarp-telemetry/pull/168

S-Dafarra, 2022-04-28:
> Not sure if it can be relevant (and even if it is, we can totally go for std::any for now): https://github.com/stephane-caron/palimpsest . See also section https://github.com/stephane-caron/palimpsest#alternatives .

Interesting! If I understood correctly, it is more like a container that can hold different types. In some sense, it is more comparable to ``matioCpp::Struct`` than ``std::any``. It could have been useful to store ``Buffer``s templated with different types, while in https://github.com/robotology/yarp-telemetry/pull/168 I worked it around by removing the use of templates from ``Buffer``.