robotology/robometry #81: Run examples as tests and under valgrind

traversaro, 2021-02-17:
Just to check if the work fine.

Nicogene, 2021-02-18:
I enabled on a branch(see https://github.com/robotology-playground/yarp-telemetry/tree/feat/addValgrind) the Valgrind tests on Linux and macOs and here is the first error:
https://github.com/robotology-playground/yarp-telemetry/runs/1925854378?check_suite_focus=true

traversaro, 2021-02-18:
This may be a false positive due to the use of a detached thread, but I am not sure.

Nicogene, 2021-02-18:
In any case, I removed the usage of the `unique_ptr` that seems to be useless in this context

https://github.com/robotology-playground/yarp-telemetry/commit/d59127aede3ba4cf0ea47dbb4ffcfb4bb56ff40f

Nicogene, 2021-02-18:
This https://github.com/robotology-playground/yarp-telemetry/commit/bda6eb2924ac1107b889fd7a6da0e763784c4a1f commit made valgrind happy.

Honestly I thought that 

`thread->detach()`

and 

`thread->join()`(at the end)

was the same.

Ref: https://stackoverflow.com/a/5612424

Nicogene, 2021-02-22:
Closed by #82

closed 2021-02-22 by Nicogene