robotology/robometry #87: Provide an example of using telemetryDeviceDumper with a file size duration of realistic size (10/15 minutes)

traversaro, 2021-03-02:


Nicogene, 2021-03-12:
Today we ran on `iCubErzelli02` the `yarprobotinterface` with the `telemetryDeviceDumper`.

We spotted some issues:
- The save thread block the closure of the `yarprobotinterface`(see #94)
- The buffer of channels were not resized after the configuration (see #93)
- If the folder where you want to create the file is not existing `matioCpp` does not create it and it fails. It actually not an issue per-se, but we have probably warn in the configuration phase that the BM will not save anything,

We dumped the accelerations, velocities and encoders of **all** the joints and here is the zip of the folder:

[test_telemetry_12min.zip](https://github.com/robotology-playground/yarp-telemetry/files/6129675/test_telemetry_12min.zip)

It is roughly 60 MB of data for 12 minutes of the experiment. We have then a rough estimate

We checked also `htop`, and the RAM consumed on `icub-head` running the yarprobotinterface was constant to ~400 MB over 8GB.

cc @traversaro @S-Dafarra @drdanz

drdanz, 2021-03-12:
> If the folder where you want to create the file is not existing `matioCpp` does not create it and it fails. It actually not an issue per-se, but we have probably warn in the configuration phase that the BM will not save anything(#95).

I'd just create the folder, and fail if the folder cannot be created... Nobody reads warnings. 😸

traversaro, 2021-03-12:
> It is roughly 60 MB of data for 12 minutes of the experiment. We have then a rough estimate

Cool! I think this gives us a nice indication.

> I'd just create the folder, and fail if the folder cannot be created... Nobody reads warnings. 😸

I totally agree.

S-Dafarra, 2021-03-12:
> I'd just create the folder, and fail if the folder cannot be created... Nobody reads warnings.

In ``matio-cpp`` unfortunately there is no way to create a folder. We use ``c++14`` there and I would avoid adding a dependency just to create a folder. Right now, ``matio-cpp`` should print an error if the file failed to open https://github.com/dic-iit/matio-cpp/blob/0f1cc62534f331c5db6e61b89c3ce7d071310567/src/File.cpp#L189-L192. In addition, it should print an error every time you try to write to that file https://github.com/dic-iit/matio-cpp/blob/0f1cc62534f331c5db6e61b89c3ce7d071310567/src/File.cpp#L304-L308

The ``isOpen`` method is public, so it can be used to check if the file got created or not.

In case there were none of those errors, that would be a little more tricky because it means that it silently failed somewhere in the middle and I would need to debug :grin:

Nicogene, 2021-03-12:
Matiocpp raised the error but since you run with the robotinterface you may miss that error and you figure out only after the end of the experiment that it didn't save anything

Nicogene, 2021-03-12:
More in details here is the content of one of the mat file I posted before

```
test_telemetry = 

  struct with fields:

    description_list: {32×1 cell}
        acceleration: [1×1 struct]
            velocity: [1×1 struct]
            encoders: [1×1 struct]
```

```
test_telemetry.encoders

ans = 

  struct with fields:

          data: [32×1×11909 double]
    dimensions: [32 1 11909]
          name: 'encoders'
    timestamps: [1×11910 double]

```
```
test_telemetry.velocity

ans = 

  struct with fields:

          data: [32×1×11909 double]
    dimensions: [32 1 11909]
          name: 'velocity'
    timestamps: [1×11910 double]

```
```
test_telemetry.acceleration

ans = 

  struct with fields:

          data: [32×1×11909 double]
    dimensions: [32 1 11909]
          name: 'acceleration'
    timestamps: [1×11910 double]
```
```
test_telemetry.description_list

ans =

  32×1 cell array

    {'torso_pitch'     }
    {'torso_roll'      }
    {'torso_yaw'       }
    {'neck_pitch'      }
    {'neck_roll'       }
    {'neck_yaw'        }
    {'l_shoulder_pitch'}
    {'l_shoulder_roll' }
    {'l_shoulder_yaw'  }
    {'l_elbow'         }
    {'l_wrist_prosup'  }
    {'l_wrist_pitch'   }
    {'l_wrist_yaw'     }
    {'r_shoulder_pitch'}
    {'r_shoulder_roll' }
    {'r_shoulder_yaw'  }
    {'r_elbow'         }
    {'r_wrist_prosup'  }
    {'r_wrist_pitch'   }
    {'r_wrist_yaw'     }
    {'l_hip_pitch'     }
    {'l_hip_roll'      }
    {'l_hip_yaw'       }
    {'l_knee'          }
    {'l_ankle_pitch'   }
    {'l_ankle_roll'    }
    {'r_hip_pitch'     }
    {'r_hip_roll'      }
    {'r_hip_yaw'       }
    {'r_knee'          }
    {'r_ankle_pitch'   }
    {'r_ankle_roll'    }
```

traversaro, 2021-03-15:
As suggested by @S-Dafarra , we could try to plot the data and check if we are not saving garbage.

Nicogene, 2021-03-17:
Today I loaded one of the 6 .mat files during the `demoYoga` and I needed this set of commands for plotting the data:
```MATLAB
t  = test_telemetry.encoders.timestamps;
t = t(:,1:end-1); % There was an extra element respect to the data
t = t-t(1,1);
joint_names = test_telemetry.description_list;
enc  = test_telemetry.encoders.data;
enc = squeeze(enc); % This for removing the dimension == 1
enc = enc'; % It was 32x11909 and we needed 11909x32
plot(t,enc);
legend(joint_names)
```
Similar steps have been applied also to `acceleartion` and `velocity` fields. And here are the graphs:

## ENCODERS
[image]

## VELOCITY - BROKEN
[image]

## VELOCITY - FIXED
I just set `ylim`
[image]

## ACCELERATION
[image]

## Conclusion

As we expected probably some data is corrupted(see the velocity plot) for the mutual exclusion not yet guaranteed. The same thing for the number of timestamps ~= to the number of samples, but I expect that they will be fixed with a proper mutual exclusion.

Moreover, we may think to change the matio structures used for avoiding these steps:
```
enc = squeeze(enc); % This for removing the dimension == 1
enc = enc'; % It was 32x11909 and we needed 11909x32
```

In general I found quite easy to use this kind of mat file for plotting the data

cc @S-Dafarra @traversaro @AlexAntn @drdanz

closed 2021-03-17 by Nicogene

S-Dafarra, 2021-03-17:
> Moreover, we may think to change the matio structures used for avoiding these steps:
> 
> ```
> enc = squeeze(enc); % This for removing the dimension == 1
> enc = enc'; % It was 32x11909 and we needed 11909x32
> ```

Regarding the second point, I think it is a matter of convention. Right now, we concatenate along the n+1 dimension, while we could concatenate on the first. I guess there are pros and cons for both cases. Maybe this can be a configuration parameter and we can also save this information in the struct.

traversaro, 2021-03-17:
The accelerometer output is also fishy, it seems way to clean and identical to the encoder output.

Nicogene, 2021-03-17:
>The accelerometer output is also fishy, it seems way to clean and identical to the encoder output.

Yeah nice catch, it is actually a bug:

https://github.com/robotology-playground/yarp-telemetry/blob/a217f3791fd8413e7d4467c7c82d1f0097910022/src/telemetryDeviceDumper/TelemetryDeviceDumper.cpp#L315