robotology/robometry #95: Warn if the path passed does not exist

Nicogene, 2021-03-12:
Since #91 it is possible to specify a path in the configuration of the BM.
`MatioCpp` does not create folders, then if the path does not exist the files will be not saved.
We should warn in the configuration phase that the path does not exist.

Nicogene, 2021-04-02:
Proposal of @drdanz -> https://github.com/robotology-playground/yarp-telemetry/issues/87#issuecomment-797463423

Nicogene, 2021-04-02:
I tried with the wrong path and you get:

```
[ERROR][matioCpp::File::Create] Failed to open the file.
[ERROR][matioCpp::File::write] The file is not open.
Failed to open /home/icub/test_telemetry/bufferConfigtest_telemetry.json
```

Without closing the yarprobotinterface, maybe it is enough and no changes are required?

S-Dafarra, 2021-04-02:
> @S-Dafarra maybe if `matio-cpp` warn which is the path of the file would be super

Sorry, I did not understand. Can you give an example of the expected behavior?

> I can open an issue if you agree

Sounds good!

Nicogene, 2021-04-02:
>Sorry, I did not understand. Can you give an example of the expected behavior?

Something like:
```
[ERROR][matioCpp::File::Create] Failed to open the file /home/icub/foo/foo.mat.
[ERROR][matioCpp::File::write] The file is not open /home/icub/foo/foo.mat.
```

I will mention it in the issue

Nicogene, 2021-04-02:
Issue opened https://github.com/dic-iit/matio-cpp/issues/48

Nicogene, 2021-04-06:
This is already done by https://github.com/dic-iit/matio-cpp/pull/49, closing

closed 2021-04-06 by Nicogene