robotology/robometry #107: telemetryDeviceDumper: problems in some .mat files saved

Nicogene, 2021-03-25:
In #87 we tried to dump some quantities for the first time using `telemetryDeviceDumper` and we obtained this list of files:

[test_telemetry_old.zip](https://github.com/robotology-playground/yarp-telemetry/files/6203985/test_telemetry_old.zip)

We found some issues and they have been fixed in #105 #104 #103 . When I analyzed the data I probably forgot to try to load all the data.
If I try to load all the data I get a problem for the first one 
```
Error using load
Cannot read file C:\Users\ngenesio\Downloads\test_telemetry\test_telemetry_1615543915.280656.mat.
```

@AlexAntn today tried again to run the same experiment(`demoYoga`) and dumping with the latest `tdd` and he got this dataset

[test_telemetry_new.zip](https://github.com/robotology-playground/yarp-telemetry/files/6203996/test_telemetry_new.zip)

And for this run I get errors in loading the **first two** 

```
>> load('test_telemetry_1616666181.080718.mat')
Error using load
Cannot read file C:\Users\ngenesio\Downloads\test_telemetry\test_telemetry_1616666181.080718.mat.
 
>> load('test_telemetry_1616666240.701385.mat')
Error using load
Cannot read file C:\Users\ngenesio\Downloads\test_telemetry\test_telemetry_1616666240.701385.mat.
```

Probably they are corrupted? 
Any Idea @S-Dafarra on how to check what is inside a mat file without loading through matlab/octave?

Nicogene, 2021-03-25:
Hold on, the new data has to be retake, we will update this issue soon!

S-Dafarra, 2021-03-25:
Just brainstorming:
- I suppose that the third file works correctly, right?
- Can you try to avoid adding a point in the file name? Just to be sure that it is not a weird problem related to the detection of the extension.
- Are you storing strings with UTF8 chars? There is an issue on previous ``matio`` versions: https://github.com/tbeu/matio/issues/164
- Can you add an assert like ``assert(linearMatrix.size() == (buffInfo.m_dimensions[0] * buffInfo.m_dimensions[1] * num_timesteps)`` https://github.com/robotology-playground/yarp-telemetry/blob/327cc2901b266198b72d3f5b1ba0a5d3edec0686/src/libYARP_telemetry/src/yarp/telemetry/BufferManager.h#L331 to make sure that we are not reading garbage?

Nicogene, 2021-03-25:
Actually, we forgot to install and we used the old version of the device(without the recent changes that fixed the concurrencies)
Now we have a different problem #110

Nicogene, 2021-03-26:
[test_telemetry_fixed.zip](https://github.com/robotology-playground/yarp-telemetry/files/6210617/test_telemetry_fixed.zip)

These are 4 files saved by the `tdd` fixed (#111). These have been taken attaching to the icub gazebo control boards. The robot was not moving, this is why velocities and acceleration are 0.
Anyhow, I can load all files correctly and now are all ~ of the same size as expected(~ 9 MB for 2 min of acquisition) except for the last one that has been save with what we had, on closure.

With the previous version of `tdd` the first files were smaller (~ 2MB) and probably corrupted for #110 

I think we can say that this have been fixed, closing

closed 2021-03-26 by Nicogene