robotology/robometry #113: Permit to log data in telemetryDeviceDumper exposed by other controlBoardInterfaces

traversaro, 2021-03-29:
For backward compatibility, it would be great to match what we support in https://robotology.github.io/robotology-documentation/doc/html/group__icub__controlBoardDumper.html .

Nicogene, 2021-05-05:
Today I tested on `iCubErzelli02` the `telemetryDeviceDumper` with #125  changes.

At the very beginning it crashes several times for mispelled names, this has been fixed here https://github.com/robotology/yarp-telemetry/pull/125/commits/abe2ea45184ecc53fd88d947821f48a88fb281a6.

Then I tried the option `logAllQuantities` and from the log of the `yarprobotinterface` I got 
```
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl
[ERROR] getPidReferenceRaw VOCAB_PIDTYPE_TORQUE  is not yet implemented for embObjMotionControl

```
Here is the resulting mat file (I almost immediately closed the `yarprobotinterface`)
[test_telemetry_with_trqPiderror.zip](https://github.com/robotology/yarp-telemetry/files/6427162/test_telemetry_with_trqPiderror.zip)

The problem here is that the view of the interface does not fail (`IPidControl`) but one of the calls is not supported, in this way disabling the `logIPidControl` for the `torque_reference` errors we have to sacrifice also `position_reference`, `position_error`, `torque_error`

traversaro, 2021-05-05:
Note that even if `position_reference`, `position_error` and `torque_error` can be read, we may need to be sure that is not creating a lot of RPC traffic between the icub-head and the EMS boards. 
See for example this code: 
* https://github.com/robotology/icub-main/blob/26c74355c9d6463762352f41f8f1a1df593844ca/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L1706
* https://github.com/robotology/icub-main/blob/26c74355c9d6463762352f41f8f1a1df593844ca/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L1502

Perhaps we can chat with @marcoaccame on this.

Nicogene, 2021-05-05:
Here 
[test_telemetry_fast_all_without_pid.zip](https://github.com/robotology/yarp-telemetry/files/6427279/test_telemetry_fast_all_without_pid.zip)

I just logged some seconds of data(all except PID quantities) and the mat file seems ok

```
test_telemetry = 

  struct with fields:

      description_list: {32×1 cell}
        motor_velocity: [1×1 struct]
          control_mode: [1×1 struct]
                torque: [1×1 struct]
               current: [1×1 struct]
                   pwm: [1×1 struct]
    motor_acceleration: [1×1 struct]
          acceleration: [1×1 struct]
         motor_encoder: [1×1 struct]
              velocity: [1×1 struct]
              encoders: [1×1 struct]
```

```

>> test_telemetry.control_mode

ans = 

  struct with fields:

          data: [32×1×1706 double]
    dimensions: [32 1 1706]
          name: 'control_mode'
    timestamps: [1×1706 double]

>> test_telemetry.pwm

ans = 

  struct with fields:

          data: [32×1×1706 double]
    dimensions: [32 1 1706]
          name: 'pwm'
    timestamps: [1×1706 double]

```

(Interaction mode is missing because I misspelled the name in the XML file)

Nicogene, 2021-05-05:
This instead it is what it is inside the .mat file of the acquisition with torque reference (`logallQuantities` enabled)
```
test_telemetry = 

  struct with fields:

      description_list: {32×1 cell}
      interaction_mode: [1×1 struct]
          control_mode: [1×1 struct]
              encoders: [1×1 struct]
              velocity: [1×1 struct]
         motor_encoder: [1×1 struct]
          acceleration: [1×1 struct]
        position_error: [1×1 struct]
    position_reference: [1×1 struct]
                torque: [1×1 struct]
          torque_error: [1×1 struct]
      torque_reference: [1×1 struct]
    motor_acceleration: [1×1 struct]
                   pwm: [1×1 struct]
               current: [1×1 struct]
        motor_velocity: [1×1 struct]
```

In particular `torque_reference` has same number of samples of the others:
```
>> test_telemetry.torque_reference

ans = 

  struct with fields:

          data: [32×1×911 double]
    dimensions: [32 1 911]
          name: 'torque_reference'
    timestamps: [1×911 double]
```
and we have only zeros 
[image]

Probably from the initialization of the vector

traversaro, 2021-05-05:
How are you moving the robot during the experiment? Unless you are controlling it in `VOCAB_CM_TORQUE_CONTROL` mode, then I think it is kind of expected that the torque references are set to 0 .

Nicogene, 2021-05-05:
> How are you moving the robot during the experiment? Unless you are controlling it in VOCAB_CM_TORQUE_CONTROL mode, then I think it is kind of expected that the torque references are set to 0 .

Nicogene, 2021-05-05:
[test_telemetry_allExecptPid_yoga_pt1.zip](https://github.com/robotology/yarp-telemetry/files/6427701/test_telemetry_allExecptPid_yoga_pt1.zip)
[test_telemetry_allExecptPid_yoga_pt2.zip](https://github.com/robotology/yarp-telemetry/files/6427702/test_telemetry_allExecptPid_yoga_pt2.zip)
[test_telemetry_allExecptPid_yoga_pt3.zip](https://github.com/robotology/yarp-telemetry/files/6427705/test_telemetry_allExecptPid_yoga_pt3.zip)
[test_telemetry_allExecptPid_yoga_pt4.zip](https://github.com/robotology/yarp-telemetry/files/6427707/test_telemetry_allExecptPid_yoga_pt4.zip)
[test_telemetry_allExecptPid_yoga_pt5.zip](https://github.com/robotology/yarp-telemetry/files/6427708/test_telemetry_allExecptPid_yoga_pt5.zip)
[test_telemetry_allExecptPid_yoga_pt6.zip](https://github.com/robotology/yarp-telemetry/files/6427710/test_telemetry_allExecptPid_yoga_pt6.zip)

This is the data of all quantities (except of PID) logged from 32 joints for ~ 12 min.
One file that contains **2 minutes of logging is ~35MB**.

commit 3fed4cbe (2021-04-29): telemetryDeviceDumper: added all the controlBoardDumper interfaces

It fixes #113
There are still open points, for now it simply dump everythin, we should define how to
group these quantities under options.

TODO

- [ ] Define proper options
- [ ] Think about the possibility of using more than one BufferManager
- [ ] Think if it is better to just convert CM and IM to doubles
- [ ] Write documentation
- [ ] Test
PR #125 (2021-05-06): telemetryDeviceDumper: added all the controlBoardDumper interfaces
It fixes #113
There are still open points, for now it simply dump everythin, we should define how to
group these quantities under options.

TODO

- [x] Define proper options
- [x] Think about the possibility of using more than one BufferManager
- [x] Think if it is better to just convert CM and IM to doubles
- [x] Write documentation
- [x] Test
- [ ] BONUS: get rid of copy-paste code
closed 2021-05-06 by Nicogene