robotology/robometry #114: Release 0.1.0 and add the repo to the robotology-superbuild

traversaro, 2021-03-29:
We should clarify in the README the expected API breakage policy (so that we may still break the API). Consider adding `experimental`  to namespace or YARP device name, as done upstream in YARP with libYARP_robotinterface.

traversaro, 2021-03-29:
Given that this is kind of expdrimental, it depens on matio-cpp and it is mostly tested on Ubutnu 20.04, a possible strategy is to add it for now just to `ROBOTOLOGY_ENABLE_DYNAMICS`  component.

Nicogene, 2021-04-01:
The statement has been added in #117

Nicogene, 2021-04-02:
The repo has been moved to robotology by @traversaro.

I released then the `v0.1.0` version -> https://github.com/robotology/yarp-telemetry/releases/tag/v0.1.0.

We are ready to add this component to the superbuild.

Nicogene, 2021-04-06:
`YARP_telemetry` has been added to the superbuild dynamics profile here -> https://github.com/robotology/robotology-superbuild/pull/677

Nicogene, 2021-04-13:
The PR has been merged, closing.

closed 2021-04-13 by Nicogene

traversaro, 2021-04-13:
Thanks to PR https://github.com/robotology/robotology-superbuild/pull/677 and the weekly build of conda packages (https://github.com/robotology/robotology-superbuild/actions/runs/743021229), we now also automatically have the yarp-telemetry conda package (see https://anaconda.org/robotology/yarp-telemetry). This means that yarp-telemetry can be installed via conda on Windows/macOS/Linux with:
~~~
conda install -c conda-forge -c robotology yarp-telemetry 
~~~

See https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md for more info.

Nicogene, 2021-04-13:
I added the conda instructions to the README : https://github.com/robotology/yarp-telemetry/commit/243bd02c38d8c034542346074cab65e79d1d13fa

traversaro, 2021-04-13:
> 
> 
> I added the conda instructions to the README : [243bd02](https://github.com/robotology/yarp-telemetry/commit/243bd02c38d8c034542346074cab65e79d1d13fa)

I think you are missing the `-c conda-forge` part of the command.