robotology/robometry #115: Try to enable compression in .mat file save

traversaro, 2021-03-29:
As suggested by @S-Dafarra .

Nicogene, 2021-04-02:
Today I tried on my laptop the `telemetryDeviceDumper` after #118 then **without** the compression, attaching to the gazebo icub simulator running `demoYoga`.

Here is the data:

[test_telemetry_preV0.1.0pt1.zip](https://github.com/robotology-playground/yarp-telemetry/files/6248801/test_telemetry_preV0.1.0pt1.zip) 

[test_telemetry_preV0.1.0pt2.zip](https://github.com/robotology-playground/yarp-telemetry/files/6248796/test_telemetry_preV0.1.0pt2.zip)

[test_telemetry_preV0.1.0pt3.zip](https://github.com/robotology-playground/yarp-telemetry/files/6248799/test_telemetry_preV0.1.0pt3.zip)

It is divided for the limitations in the file size of GH.

This will be our term of comparison while we will enable the compression

Nicogene, 2021-04-19:
Here https://github.com/robotology/yarp-telemetry/issues/116#issuecomment-822292398 and here https://github.com/robotology/yarp-telemetry/issues/116#issuecomment-822295725 we have the performances of saving buffers of integers **without** compression

Nicogene, 2021-04-19:
This is the impact of the compression in the benchmarching (always in `Release`)
```
-------------------------------------------------------------------------------
Buffer Manager Test
  Benchmarching section scalar int
-------------------------------------------------------------------------------
/home/ngenesio/robotology/yarp-telemetry/test/BufferManagerTest.cpp:261
...............................................................................

benchmark name                       samples       iterations    estimated
                                     mean          low mean      high mean
                                     std dev       low std dev   high std dev
-------------------------------------------------------------------------------
BufferOfInt-1000Samples-
oneVariable-1x1                                100            24      2.196 ms
                                         23.289 us    22.0523 us    29.0057 us
                                        11.5012 us    1.30895 us    27.2576 us

BufferOfInt-10000Samples-
oneVariable-1x1                                100            25      2.205 ms
                                        189.423 us    188.531 us     190.38 us
                                        4.71909 us    4.00428 us    5.67096 us

BufferOfInt-100000Samples-
oneVariable-1x1                                100            14     2.2904 ms
                                        3.84534 ms    3.82939 ms    3.88533 ms
                                        121.131 us    59.8298 us    254.113 us

-------------------------------------------------------------------------------
Buffer Manager Test
  Benchmarching section vector int
-------------------------------------------------------------------------------
/home/ngenesio/robotology/yarp-telemetry/test/BufferManagerTest.cpp:303
...............................................................................

benchmark name                       samples       iterations    estimated
                                     mean          low mean      high mean
                                     std dev       low std dev   high std dev
-------------------------------------------------------------------------------
BufferOfInt-1000Samples-
oneVariable-3x1                                100            26      2.184 ms
                                        23.7464 us    23.5317 us    24.0815 us
                                        1.34468 us    939.276 ns    1.90603 us

BufferOfInt-10000Samples-
oneVariable-3x1                                100            25     2.1775 ms
                                        210.662 us    209.416 us    212.311 us
                                        7.22334 us    5.66413 us    11.2294 us

BufferOfInt-100000Samples-
oneVariable-3x1                                100            14     2.3044 ms
                                        3.78658 ms    3.77177 ms    3.80836 ms
                                        90.1273 us    66.2636 us     151.99 us

-------------------------------------------------------------------------------
Buffer Manager Test
  Benchmarching section matrix int
-------------------------------------------------------------------------------
/home/ngenesio/robotology/yarp-telemetry/test/BufferManagerTest.cpp:350
...............................................................................

benchmark name                       samples       iterations    estimated
                                     mean          low mean      high mean
                                     std dev       low std dev   high std dev
-------------------------------------------------------------------------------
BufferOfInt-1000Samples-
oneVariable-3x2                                100            25     2.1725 ms
                                        27.0848 us    26.6739 us    27.8695 us
                                          2.792 us    1.71097 us    4.27228 us

[... 67 lines cut ...]
Buffer Manager Test
  Benchmarching section matrix int
-------------------------------------------------------------------------------
/home/ngenesio/robotology/yarp-telemetry/test/BufferManagerTest.cpp:364
...............................................................................

benchmark name                       samples       iterations    estimated
                                     mean          low mean      high mean
                                     std dev       low std dev   high std dev
-------------------------------------------------------------------------------
BufferOfInt-1000Samples-
oneVariable-3x2                                100            22     2.2704 ms
                                        7.54132 us    7.31777 us    8.07836 us
                                        1.66523 us    807.941 ns    3.28624 us

BufferOfInt-10000Samples-
oneVariable-3x2                                100            26     2.2932 ms
                                        22.2203 us    21.9476 us    22.6629 us
                                        1.74048 us    1.17876 us    2.40411 us

BufferOfInt-100000Samples-
oneVariable-3x2                                100            25      2.285 ms
                                        264.646 us    260.733 us    268.992 us
                                        20.9638 us    18.0406 us    26.2273 us
```

Nicogene, 2021-04-20:
I tried the same experiment of https://github.com/robotology/yarp-telemetry/issues/115#issuecomment-812452312 after enabling the **compression** here is the data:

[test_telemetry_compressed_pt1.zip](https://github.com/robotology/yarp-telemetry/files/6343519/test_telemetry_compressed_pt1.zip)
[test_telemetry_compressed_pt2.zip](https://github.com/robotology/yarp-telemetry/files/6343517/test_telemetry_compressed_pt2.zip)
[test_telemetry_compressed_pt3.zip](https://github.com/robotology/yarp-telemetry/files/6343518/test_telemetry_compressed_pt3.zip)

Here is the data, it is ok there are no artifacts:
[image]

For 2 minutes of acquisition of encoders/accelerations/velocities of 32 joints we have:

~ 9.2 MB uncompressed
~ 5.7 MB compressed

The differences in computation seem more significant, we have to decide if we want to lose something in performances for saving some space.

cc @AlexAntn @traversaro @S-Dafarra @drdanz

S-Dafarra, 2021-04-20:
Nice! Would it be possible to save in another thread without blocking the circular buffer?

drdanz, 2021-04-22:
Since the impact is rather significant, I believe that it should be a user choice, we could have an option to select is compression should be enabled or not...

PR #123 (2021-04-28): BufferManager: add zlib compression
This requires `matioCpp v0.1.1`.

To be merged after #121 

It fixes #115
closed 2021-04-28 by Nicogene