robotology/robometry #116: Add a test for benchmarking the time spent in copying buffer when file is saved

traversaro, 2021-03-29:
We could use https://github.com/catchorg/Catch2/blob/devel/docs/benchmarks.md to measure this.

Nicogene, 2021-04-19:
In #121 the benchmarking has been added in the unit tests.

The test are for 1K/10K sample for integers variables 1x1/3x1/3x2.
Here are the results run in `debug`:
```
-------------------------------------------------------------------------------
Buffer Manager Test
  Benchmarching section scalar int
-------------------------------------------------------------------------------
..\..\test\BufferManagerTest.cpp(261)
...............................................................................

benchmark name                       samples       iterations    estimated
                                     mean          low mean      high mean
                                     std dev       low std dev   high std dev
-------------------------------------------------------------------------------
BufferOfInt-1000Samples-
oneVariable-1x1                                100             8    17.7128 ms
                                        376.529 us    364.241 us    395.902 us
                                        77.7126 us    55.7227 us     124.77 us

BufferOfInt-10000Samples-
oneVariable-1x1                                100             8    18.3664 ms
                                        1.16241 ms    1.15065 ms    1.18463 ms
                                        80.0399 us    50.5456 us    150.749 us

-------------------------------------------------------------------------------
Buffer Manager Test
  Benchmarching section vector int
-------------------------------------------------------------------------------
..\..\test\BufferManagerTest.cpp(310)
...............................................................................

benchmark name                       samples       iterations    estimated
                                     mean          low mean      high mean
                                     std dev       low std dev   high std dev
-------------------------------------------------------------------------------
BufferOfInt-1000Samples-
oneVariable-3x1                                100             8    18.1952 ms
                                        365.821 us    358.884 us    374.361 us
                                        39.0475 us    32.1809 us    47.7844 us

BufferOfInt-10000Samples-
oneVariable-3x1                                100             7    18.3491 ms
                                        1.56583 ms    1.54305 ms    1.59681 ms
                                        134.484 us     104.13 us     174.04 us

-------------------------------------------------------------------------------
Buffer Manager Test
  Benchmarching section matrix int
-------------------------------------------------------------------------------
..\..\test\BufferManagerTest.cpp(364)
...............................................................................

benchmark name                       samples       iterations    estimated
                                     mean          low mean      high mean
                                     std dev       low std dev   high std dev
-------------------------------------------------------------------------------
BufferOfInt-1000Samples-
oneVariable-3x2                                100             8    17.9568 ms
                                        385.918 us    378.058 us    397.654 us
                                        48.4551 us    35.7654 us    73.9177 us

BufferOfInt-10000Samples-
oneVariable-3x2                                100             7    17.4748 ms
                                        1.89913 ms    1.87582 ms    1.92889 ms
                                        133.736 us    109.241 us    185.824 us
```

Nicogene, 2021-04-19:
Here are the results in `release`:
```
-------------------------------------------------------------------------------
Buffer Manager Test
  Benchmarching section scalar int
-------------------------------------------------------------------------------
/home/ngenesio/robotology/yarp-telemetry/test/BufferManagerTest.cpp:261
...............................................................................

benchmark name                       samples       iterations    estimated
                                     mean          low mean      high mean
                                     std dev       low std dev   high std dev
-------------------------------------------------------------------------------
BufferOfInt-1000Samples-
oneVariable-1x1                                100            25     2.2925 ms
                                        5.78244 us    5.65912 us    5.99624 us
                                         802.39 ns    544.518 ns    1.34566 us

BufferOfInt-10000Samples-
oneVariable-1x1                                100            26     2.3062 ms
                                        15.9213 us    15.6928 us    16.4777 us
                                        1.70486 us    881.002 ns    3.47162 us

-------------------------------------------------------------------------------
Buffer Manager Test
  Benchmarching section vector int
-------------------------------------------------------------------------------
/home/ngenesio/robotology/yarp-telemetry/test/BufferManagerTest.cpp:310
...............................................................................

benchmark name                       samples       iterations    estimated
                                     mean          low mean      high mean
                                     std dev       low std dev   high std dev
-------------------------------------------------------------------------------
BufferOfInt-1000Samples-
oneVariable-3x1                                100            27     2.2788 ms
                                        5.80815 us    5.70567 us    5.95907 us
                                        628.314 ns    463.723 ns    918.706 ns

BufferOfInt-10000Samples-
oneVariable-3x1                                100            27     2.3004 ms
                                        17.8149 us    17.4488 us    18.5954 us
                                        2.60438 us    1.30532 us    4.38984 us

-------------------------------------------------------------------------------
Buffer Manager Test
  Benchmarching section matrix int
-------------------------------------------------------------------------------
/home/ngenesio/robotology/yarp-telemetry/test/BufferManagerTest.cpp:364
...............................................................................

benchmark name                       samples       iterations    estimated
                                     mean          low mean      high mean
                                     std dev       low std dev   high std dev
-------------------------------------------------------------------------------
BufferOfInt-1000Samples-
oneVariable-3x2                                100            27     2.3382 ms
                                        8.89085 us    7.95481 us     10.934 us
                                        6.68572 us    3.85793 us    12.3458 us

BufferOfInt-10000Samples-
oneVariable-3x2                                100            27     2.3166 ms
                                        22.0219 us    21.7198 us    22.7437 us
                                        2.24315 us    1.16811 us    4.48527 us
```

PR #121 (2021-04-20): Tests: add benchmarking for integers
Added the benchmarking section in `BufferManagerTest`.

It can be enable with `ENABLE_BENCHMARKING` cmake option.

TODO: remove https://github.com/robotology/yarp-telemetry/pull/121/commits/41e53d06239735337c338c22209a68539f9541c1 before merging, it has been added because the messages printed ruin the format of the benchmarking

It fixes #116 

cc @S-Dafarra @drdanz
closed 2021-04-20 by Nicogene