robotology/robometry #119: Failed to load plugin. Cannot find or load shared library

S-Dafarra, 2021-04-09:
I have installed ``yarp-telemetry`` on the iCub3 head and I managed to compile with no major issue. I have installed it in the folder 
```
/usr/local/src/robot/yarp-telemetry/build/install
```
and edited the ``.bashrc`` as follows
```
export YARP_DATA_DIRS=${YARP_DATA_DIRS}:${ROBOT_CODE}/yarp-telemetry/build/install/share/yarp
export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:${ROBOT_CODE}/yarp-telemetry/build/install/lib
```
but if I launch ``yarp plugin telemetryDeviceDumper``, I always get the following error, independently from where I launch it:

```
icub@icub-head:/usr/local/src/robot/yarp-telemetry/build/install/share/yarp(master)$ yarp plugin telemetryDeviceDumper
Yes, this is a YARP plugin
[ERROR]     Cannot find or load shared library
```

I wonder what I am doing wrong.

cc @Nicogene @AlexAntn @traversaro

Nicogene, 2021-04-12:
I usually install everything in the install folder of the superbuild and unfortunately this has hidden this error.

The problem does not occur when I install in the superbuild  install folder:

```bash
$ ldd /home/ngenesio/robotology/robotology-superbuild/build/install/lib/yarp/yarp_telemetryDeviceDumper.so
        linux-vdso.so.1 (0x00007ffdeb383000)
        libYARP_dev.so.3 => /home/ngenesio/robotology/robotology-superbuild/build/install/lib/libYARP_dev.so.3 (0x00007f7a8d81f000)
        libYARP_telemetry.so.0 => /home/ngenesio/robotology/robotology-superbuild/build/install/lib/libYARP_telemetry.so.0 (0x00007f7a8d740000)
        libYARP_os.so.3 => /home/ngenesio/robotology/robotology-superbuild/build/install/lib/libYARP_os.so.3 (0x00007f7a8d569000)
        libmatioCpp.so.0.1.0 => /home/ngenesio/robotology/robotology-superbuild/build/install/lib/libmatioCpp.so.0.1.0 (0x00007f7a8d545000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f7a8d4fd000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f7a8d31a000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f7a8d2ff000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f7a8d10d000)
        libYARP_math.so.3 => /home/ngenesio/robotology/robotology-superbuild/build/install/lib/libYARP_math.so.3 (0x00007f7a8d0c0000)
        libYARP_sig.so.3 => /home/ngenesio/robotology/robotology-superbuild/build/install/lib/libYARP_sig.so.3 (0x00007f7a8d07a000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f7a8cf29000)
        libACE-6.4.5.so => /usr/lib/libACE-6.4.5.so (0x00007f7a8cd94000)
        libedit.so.2 => /usr/lib/x86_64-linux-gnu/libedit.so.2 (0x00007f7a8cd5c000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f7a8d975000)
        libmatio.so.9 => /usr/lib/x86_64-linux-gnu/libmatio.so.9 (0x00007f7a8ccfb000)
        libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f7a8cc76000)
        libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f7a8cc3c000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f7a8cc36000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f7a8cc2b000)
        libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007f7a8cbfb000)
        libbsd.so.0 => /usr/lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f7a8cbe1000)
        libhdf5_serial.so.103 => /usr/lib/x86_64-linux-gnu/libhdf5_serial.so.103 (0x00007f7a8c864000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f7a8c846000)
        libsz.so.2 => /usr/lib/x86_64-linux-gnu/libsz.so.2 (0x00007f7a8c841000)
        libaec.so.0 => /usr/lib/x86_64-linux-gnu/libaec.so.0 (0x00007f7a8c838000)
```

On the other hand here is the output of ldd of the plugin if installed in the folder used by @S-Dafarra 
```bash
$ ldd /home/ngenesio/robotology/yarp-telemetry/build/install/lib/yarp/yarp_telemetryDeviceDumper.so
 linux-vdso.so.1 (0x00007ffd767bf000)
        libYARP_dev.so.3 => /home/ngenesio/robotology/robotology-superbuild/build/install/lib/libYARP_dev.so.3 (0x00007f6093e09000)
        libYARP_telemetry.so.0 => not found
        libYARP_os.so.3 => /home/ngenesio/robotology/robotology-superbuild/build/install/lib/libYARP_os.so.3 (0x00007f6093c0d000)
        libmatioCpp.so.0.1.0 => /home/ngenesio/robotology/robotology-superbuild/build/install/lib/libmatioCpp.so.0.1.0 (0x00007f6093be9000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f6093bc4000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f60939e3000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f60939c8000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f60937d6000)
        libYARP_math.so.3 => /home/ngenesio/robotology/robotology-superbuild/build/install/lib/libYARP_math.so.3 (0x00007f6093789000)
        libYARP_sig.so.3 => /home/ngenesio/robotology/robotology-superbuild/build/install/lib/libYARP_sig.so.3 (0x00007f6093743000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f60935f2000)
        libACE-6.4.5.so => /usr/lib/libACE-6.4.5.so (0x00007f609345d000)
        libedit.so.2 => /usr/lib/x86_64-linux-gnu/libedit.so.2 (0x00007f6093425000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f6093f5f000)
        libmatio.so.9 => /usr/lib/x86_64-linux-gnu/libmatio.so.9 (0x00007f60933c4000)
        libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f609333f000)
        libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f6093305000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f60932ff000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f60932f4000)
        libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007f60932c4000)
        libbsd.so.0 => /usr/lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f60932aa000)
        libhdf5_serial.so.103 => /usr/lib/x86_64-linux-gnu/libhdf5_serial.so.103 (0x00007f6092f2d000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f6092f0f000)
        libsz.so.2 => /usr/lib/x86_64-linux-gnu/libsz.so.2 (0x00007f6092f0a000)
        libaec.so.0 => /usr/lib/x86_64-linux-gnu/libaec.so.0 (0x00007f6092f01000)

```
The problem should be ` libYARP_telemetry.so.0 => not found`

traversaro, 2021-04-12:
I guess it is a good old rpath error. Probably it can be solved by using the rpath helper from YCM

traversaro, 2021-04-12:
> 
> 
> I guess it is a good old rpath error. Probably it can be solved by using the rpath helper from YCM

https://github.com/robotology/how-to-export-cpp-library/blob/307f7bd78789cb68aa58c7804a3549ec23679483/CMakeLists.txt#L106

Nicogene, 2021-04-12:
Unfortunately, it seems that we are already setting it:

https://github.com/robotology/yarp-telemetry/blob/88b694a47d29d6100b5f66a6a1691da77d29f252/CMakeLists.txt#L39-L44

drdanz, 2021-04-12:
You should add also the plugin path to the RPATH binary dirs, for example this is taken from YARP:

```cmake
add_install_rpath_support(LIB_DIRS "${CMAKE_INSTALL_FULL_LIBDIR}"       # Libraries
                          BIN_DIRS "${CMAKE_INSTALL_FULL_BINDIR}"       # Binaries
                                   "${CMAKE_INSTALL_FULL_LIBDIR}/yarp"  # Plugins
                          INSTALL_NAME_DIR "${CMAKE_INSTALL_FULL_LIBDIR}"
                          USE_LINK_PATH)
```

commit 78434907 (2021-04-12): telemetryDeviceDumper: fix plugin load

Fixes #119
closed 2021-04-12 by Nicogene