robotology/robometry #126: Add support for odometry to telemetryDeviceDumper

drdanz, 2021-04-23:
According to @randaz81 the data he would like to dump on the R1 robot is the odometry data coming from the `ILocalization2D` interface.

Data is generated by the `odomLocalizer` device, and is published on the yarp network by `localization2DServer` (now `localization2D_nws_yarp`)

Nicogene, 2021-06-08:
The structure of `OdometryData` is:
```c++
double odom_x;

double odom_y;

double odom_theta;

double base_vel_x;

double base_vel_y;

double base_vel_theta;

double odom_vel_x;

double odom_vel_y;
```

We can encode this in 2 ways:
- 1 channel of size 8
- 8 separated channels of size 1

I personally prefer the first solution, because to the 8 channels you have to add also the others if enabled and may mess up the structure of the mat file.

What do you think about it @randaz81 @drdanz @AlexAntn @traversaro ?

randaz81, 2021-06-09:
I am also for the first solution.

From the logic point of view, another possible solution would be to have three channels, each of them consisting of three values inside (if you notice, three values are positions, three values are velocities, three values are velocities in a different coordinate frame). But I don't know if it is feasible, so the first solution is ok.

PR #138 (2021-06-15): telemetryDeviceDumper: add log of Odometry
This PR adds the possibility to log the odometry data.
It fixes #126 

There are still some open points:

- Differently from the other data retrieved through the attaching and the control_board_remapper, for the odometry we used the `localization2DClient`, since the localization device can run on different machines from the `r1-base`, and the localization pipeline has never been used with `yarprobotinterface`. We have to discuss the pro and cons of following this approach or the "attach" approach
- Now we added the log odometry to `logAllQuantities` o [... cut]
closed 2021-06-15 by Nicogene