robotology/robometry #127: yarp-telemetry does not generate a plugin.ini

drdanz, 2021-04-23:
Note: This might actually be an issue in YARP... I'll start the discussion here, and then fix it in YARP.

yarp-telemetry does not call `yarp_configure_plugins_installation` and uses instead YARP directories to install the plugins, therefore the plugin `.ini`/manifest file is not generated.
This is perfectly fine for the install tree version (assuming that it is installed in the same prefix as YARP), but does not work for the build tree version and if it is installed in a different prefix.

I think that there should be a way to generate the .ini for the build directory (so that it can be used with `YARP_DATA_DIRS`, that is not installed if the installation prefix is the same (or perhaps not installed at all). Perhaps this makes sense only if you choose to use YARP installation directories. What do you think?

drdanz, 2021-04-23:
one option could be make a special case if `YARP_CONFIGURE_PLUGINS_INSTALLATION` is called with `yarp` as component

traversaro, 2021-04-23:
If I remember well, `yarp_configure_plugins_installation` was indeed introduced to support the cases in which the repo was installing stuff in YARP directories without using `yarp_configure_external_installation` (see https://github.com/robotology/yarp/blob/5313b2defd8be7237567ea56eca8dde93dfc8121/doc/release/v2_3_70.md#cmake-modules-1 and https://github.com/robotology/yarp/pull/1111). So perhaps the solution is just to add a call to `yarp_configure_plugins_installation` ?

drdanz, 2021-04-23:
If you want to use a different name (for example `yarp_configure_plugins_installation(yarp-telemetry)`) that should just work, but if you use `yarp_configure_plugins_installation(yarp)` it will overwrite the original `yarp.ini` which is not a good behaviour. That's why I was suggesting a special case when you specify `yarp` as component

traversaro, 2021-04-23:
I think we can just use `yarp_configure_plugins_installation(yarp-telemetry)`, it would work fine without complexity both if YARP is installed in the same prefix or not.

commit b1515f93 (2021-04-23): Make yarp-telemetry generate the plugin .ini

It fixes #127
closed 2021-04-28 by Nicogene