robotology/robometry #135: Failed to compile on MacOS (filesystem headers not found)

lrapetti, 2021-05-27:
I tried to compile `yarp-telemetery`, and it is failing with the following output:
```
$ make
[  8%] Built target YARP_telemetry
[ 12%] Building CXX object src/telemetryDeviceDumper/CMakeFiles/yarp_telemetryDeviceDumper.dir/yarp_plugin_telemetryDeviceDumper.cpp.o
In file included from /Users/lorenzorapetti/Software/robotology-superbuild/build/robotology/yarp-telemetry/src/telemetryDeviceDumper/yarp_plugin_telemetryDeviceDumper.cpp:13:
In file included from /Users/lorenzorapetti/Software/robotology-superbuild/robotology/yarp-telemetry/src/telemetryDeviceDumper/TelemetryDeviceDumper.h:26:
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/yarp-telemetry/src/libYARP_telemetry/src/yarp/telemetry/experimental/BufferManager.h:40:6: error: static_assert failed "Neither <filesystem> nor <experimental/filesystem> headers are present in the system, but they are required"
     static_assert(false, "Neither <filesystem> nor <experimental/filesystem> headers are present in the system, but they are required"); 
     ^             ~~~~~
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/yarp-telemetry/src/libYARP_telemetry/src/yarp/telemetry/experimental/BufferManager.h:92:14: warning: unused variable 'ok' [-Wunused-variable]
        bool ok = configure(_bufferConfig);
             ^
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/yarp-telemetry/src/libYARP_telemetry/src/yarp/telemetry/experimental/BufferManager.h:152:46: error: use of undeclared identifier 'yarp_telemetry_fs'
        if (!m_bufferConfig.path.empty() && !yarp_telemetry_fs::exists(m_bufferConfig.path)) {
                                             ^
1 warning and 2 errors generated.
make[2]: *** [src/telemetryDeviceDumper/CMakeFiles/yarp_telemetryDeviceDumper.dir/yarp_plugin_telemetryDeviceDumper.cpp.o] Error 1
make[1]: *** [src/telemetryDeviceDumper/CMakeFiles/yarp_telemetryDeviceDumper.dir/all] Error 2
make: *** [all] Error 2
```

I am on MacOS 10.14.6 (actions are running for `macos-latest` which is `10.15`)

traversaro, 2021-05-27:
That is probably my fault: I thought 10.15 was old enough for all IIT mac users. I will do a quick poll to understand the situation.

drdanz, 2021-05-28:
If I didn't miss any occurrence, I think that the `filesystem` library is used only for `exists`, you can probably write a fake implementation using `::stat`... i.e. something like this

```c++
#  elif __has_include(<sys/stat.h>)
#    include <sys/stat.h>
    namespace yarp_telemetry_fs {
        // WARNING: Not the actual signature of the method, but in this context
        //          it is used only using string as argument
        bool exists(const std::string& p)
        {
            struct ::stat dummy;
            return (::stat(p.c_str(), &dummy) != -1);
        }
    }

```

traversaro, 2021-05-28:
Switching all CIs  also on 10.14 may reveal other issue, so unless there is any compelling reason for supporting 10.14 (I am collecting feedback) I think the solution may be to kindly ask @lrapetti to update his OS. : )

lrapetti, 2021-05-28:
> I think the solution may be to kindly ask @lrapetti to update his OS. : )

No problem for me

drdanz, 2021-06-08:
Did you manage to install this? Do you have any idea if this depends on the version of macos or on the version of xcode?

lrapetti, 2021-06-09:
I have recently switched to MacOS 11.3.1 and Xcode 12.5, and it is now compiling without errors. Not sure though which of the two solved the issue, but I think we can close it

traversaro, 2021-06-09:
I think it is the macOS SDK version. If I am not wrong, you can target an old version even with a recent Xcode version, but the code will continue to not compile if you target macOS <= 10.14 . As only one last user in the lab is still using 10.14 and all the CI run on 10.15, for me we can target 10.15 support and close the issue. @Nicogene ?

Nicogene, 2021-06-09:
> I think it is the macOS SDK version. If I am not wrong, you can target an old version even with a recent Xcode version, but the code will continue to not compile if you target macOS <= 10.14 . As [...]

Totally agree, maybe we can write something about it in the readme

closed 2021-06-09 by Nicogene