robotology/robometry #137: Missing motor information with robot in simulation

isorrentino, 2021-06-04:
I tried to use `yarp-telemetry` with the robot in simulation and I used the following config file:

```xml

<?xml version="1.0" encoding="UTF-8" ?>
<!DOCTYPE devices PUBLIC "-//YARP//DTD yarprobotinterface 3.0//EN" "http://www.yarp.it/DTD/yarprobotinterfaceV3.0.dtd">
  
<robot name="yarp_telemetry" build="1" xmlns:xi="http://www.w3.org/2001/XInclude">

  <device name="right_leg_mc" type="remote_controlboard">
    <param name="remote"> /icubSim/right_leg </param>
    <param name="local"> /yarptelemetry/right_leg </param>
  </device>

  <device xmlns:xi="http://www.w3.org/2001/XInclude" name="telemetryDeviceDumper" type="telemetryDeviceDumper">
    <param name="axesNames">(r_hip_pitch,r_hip_roll,r_hip_yaw,r_knee,r_ankle_pitch,r_ankle_roll)</param>
    <param name="logIEncoders">true</param>
    <param name="logITorqueControl">true</param>
    <param name="logIMotorEncoders">false</param>
    <param name="logIControlMode">true</param>
    <param name="logIInteractionMode">true</param>
    <param name="logIPidControl">true</param>
    <param name="logIAmplifierControl">true</param>
    <param name="saveBufferManagerConfiguration">true</param>
    <param name="experimentName">test_telemetry</param>
    <param name="path">/home/isorrentino/devel/Telexistence/Datasets/yarp_telemetry/</param>
    <param name="n_samples">100000</param>
    <param name="save_periodically">true</param>
    <param name="save_period">120.0</param>
    <param name="data_threshold">100</param>
    <param name="auto_save">true</param>
    
    <action phase="startup" level="15" type="attach">
      <paramlist name="networks">
        <!-- motorcontrol and virtual torque sensors -->
        <elem name="right_leg">right_leg_mc</elem>
        <!-- ft -->
      </paramlist>
    </action>
    
    <action phase="shutdown" level="2" type="detach" />
    
  </device>
</robot>
```

But the application prints the following warnings:

```bash
[WARNING] telemetryDeviceDumper warning : joint torque errors was not readed correctly
[WARNING] telemetryDeviceDumper warning : joint torque references was not readed correctly
[WARNING] telemetryDeviceDumper warning : voltage PWM was not readed correctly
```

I'm opening this issue as I expect to read the torque references and errors even when the robot is in simulation. Is there something I'm not considering, so the warnings are ok, or should I see those quantities also in simulation?

traversaro, 2021-06-04:
I guess the methods related to those quantities need to be implemented in `gazebo_yarp_controlboard` .

prashanthr05, 2021-06-10:
> I guess the methods related to those quantities need to be implemented in `gazebo_yarp_controlboard` .

> [WARNING] telemetryDeviceDumper warning : voltage PWM was not readed correctly

In looking at the gazebo controlboard code, the [methods](https://github.com/robotology/yarp-telemetry/blob/cb9658be72844fb28b6d1e0f444b71c145c2aed5/src/telemetryDeviceDumper/TelemetryDeviceDumper.cpp#L533) related to voltage PWM which is a part of IAmplifierControl interface does not seem to be implemented.

> [WARNING] telemetryDeviceDumper warning : joint torque references was not readed correctly

This error is not expected, since the method is implemented [here](https://github.com/robotology/gazebo-yarp-plugins/blob/0d18d2ed4e20c6cfb751e976171781df0deb9acd/plugins/controlboard/src/ControlBoardDriverPidControl.cpp#L231-L268).

> [WARNING] telemetryDeviceDumper warning : joint torque errors was not readed correctly

Instead, the errors related to the joint torque errors are strange.
Especially, with [torque errors](https://github.com/robotology/yarp-telemetry/blob/cb9658be72844fb28b6d1e0f444b71c145c2aed5/src/telemetryDeviceDumper/TelemetryDeviceDumper.cpp#L507) from the gazebo control board implementation of IPIDControl interface. When the PID type is passed as `VOCAB_CM_TORQUE`, the [method](https://github.com/robotology/gazebo-yarp-plugins/blob/0d18d2ed4e20c6cfb751e976171781df0deb9acd/plugins/controlboard/src/ControlBoardDriverPidControl.cpp#L120-L136) returning false is not expected, moreover, whatever be the PID type, the method is returning only position errors.

Nicogene, 2021-06-10:
@prashanthr05 @traversaro shall we open those 3 issues in `gazebo-yarp-plugins`?

I think that here we are just using the interfaces, I don't think that there are problems in the `telemetryDeviceDumper`, also because those methods on real `iCub` work

traversaro, 2021-06-10:
> @prashanthr05 @traversaro shall we open those 3 issues in gazebo-yarp-plugins?

Ok.

Nicogene, 2021-06-10:
Closing in favor of robotology/gazebo-yarp-plugins#554 robotology/gazebo-yarp-plugins#555 robotology/gazebo-yarp-plugins#556

closed 2021-06-10 by Nicogene