robotology/robometry #145: Request info on BufferManager usage

GiulioRomualdi, 2021-12-10:
Hi all, I am trying to use yarp telemetry library in https://github.com/ami-iit/bipedal-locomotion-framework/tree/master/devices/YarpRobotLoggerDevice and I have two questions:
1. As far as I understood the `yarp::telemetry::experimental::BufferManager` is the main class that is used to save the data in a `mat` file. I noticed that the user should specify a type for a given `BufferManager`. Now let's assume I want to store in the same `mat` files both int vector and double vectors is it possible? 
2. Back in time we were used to store the data by grouping them into "types". For instance, all the FTs belong to the same Matlab structure and then each element of the structure is a vector containing the 6 elements of the wrench over time.  
See [here](https://github.com/ami-iit/bipedal-locomotion-framework/blob/6908317ac03d678843460792f4762e95c415e881/devices/YarpRobotLoggerDevice/src/YarpRobotLoggerDevice.cpp#L412-L433). Is it possible to replicate this with yarp-telemetry? For instance, it is possible to have multiple `BufferManagers` in the same mat file? If this would be possible I can create a `BufferManager` for each "type" of data (e.g. one for `FTs`, one for `JointState`, one for the high-level info (CoM ZMP, ...))
I am afraid that this is not possible since according to the documentation there seems that for each `BufferManager` there exists one `Matlab` file.

cc @traversaro @Nicogene @S-Dafarra

GiulioRomualdi, 2021-12-10:
I'm afraid that the answers of bot the questions are no 

Since the `BufferManager` contains a `map` of `BufferInfo` of the same type

https://github.com/robotology/yarp-telemetry/blob/8a5cffe81ea65c44dd2b84c176674a8bf1466b58/src/libYARP_telemetry/src/yarp/telemetry/experimental/BufferManager.h#L473

Nicogene, 2021-12-10:
Hi @GiulioRomualdi!

You answered by yourself 😄 

As you mentioned, at this moment each `BufferManager` is associated with a single .mat file and it handles only one type.

The first point is a known limitation (see https://github.com/robotology/yarp-telemetry/issues/76) that can be addressed with some template black magic or some modern construct like `std::any`.
The second point you posted could be handled by the "deamon/collector", we discussed about the need to have something that collect all the data and merge it in one mat file and one place (https://github.com/robotology/yarp-telemetry/issues/51)

closed 2022-01-07 by Nicogene