robotology/robometry #153: Unable to include AddInstallRPATHSupport if find_package(YARP COMPONENT telemetry) has been called

GiulioRomualdi, 2021-12-17:
I am trying to use `yarp-temetery` in  [bipedal-locomotion-framework](https://github.com/ami-iit/bipedal-locomotion-framework) and I noticed that if I call
```cmake
find_package(YARP COMPONENT telemetry) 
``` 
before `include(AddInstallRPATHSupport)` I got the following  error
```
-- Found YARP: /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.5.0+101-20211025.9+git0baa202b9")
-- Found YARP: /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.5.0+101-20211025.9+git0baa202b9")
CMake Error at CMakeLists.txt:65 (include):
  include could not find load file:

    AddInstallRPATHSupport

CMake Error at CMakeLists.txt:66 (add_install_rpath_support):
  Unknown CMake command "add_install_rpath_support".

-- Configuring incomplete, errors occurred!
```

If `find_package(YARP COMPONENT telemetry) ` is called after `include(AddInstallRPATHSupport)` everything works as expected. 

I remember that back in time this problem was already faced by @S-Dafarra https://github.com/ami-iit/matio-cpp/pull/41 and https://github.com/ami-iit/bipedal-locomotion-framework/pull/253

However, in this case, `InstallBasicPackageFiles` is directly retrieved by YCM and `OVERRIDE_MODULE_PATH` does not exist in the official version of `install_basic_package_files` cmake function

cc @Nicogene @traversaro @AlexAntn

S-Dafarra, 2021-12-17:
Maybe it is exactly the same problem. Can you try switching ``matioCpp`` to ``main``?

GiulioRomualdi, 2021-12-17:
> Maybe it is exactly the same problem. Can you try switching `matioCpp` to `main`?

I've already matioCpp in master

GiulioRomualdi, 2021-12-17:
I think we should implement something similar of what is done in matioCpp also in telemetry

S-Dafarra, 2021-12-17:
Ok because the reason behind that error was due to ``matioCpp`` editing the ``CMAKE_MODULE_PATH``. Can you check the value of that variable before and after ``find_package(YARP COMPONENT telemetry)``?

GiulioRomualdi, 2021-12-17:
This is the cmake code
```cmake
list(APPEND CMAKE_MODULE_PATH ${CMAKE_CURRENT_SOURCE_DIR}/cmake)
message("--> CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}")
find_package(YARP COMPONENTS telemetry)
message("--> CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}")
```

This is the output
```
--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/robotology-superbuild/src/bipedal-locomotion-framework/cmake
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0")  
-- Found YARP: /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.5.0+101-20211025.9+git0baa202b9")
-- Found YARP: /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.5.0+101-20211025.9+git0baa202b9")
--> CMAKE_MODULE_PATH 
```
The content of CMAKE_MODULE_PATH is erased

S-Dafarra, 2021-12-17:
Can you repeat the same test by editing the ``YARP_telemetryConfig.cmake`` file that should be located in ``robotology-superbuild/build/install/lib/cmake/YARP_telemetry``? 

For example, that file looks like this to me:
```cmake
set(YARP_telemetry_VERSION 3.5.100+20211203.5+gitb7a6221dd+dirty)

####### Expanded from @PACKAGE_INIT@ by configure_package_config_file() #######
####### Any changes to this file will be overwritten by the next CMake run ####
####### The input file was YARP_telemetryConfig.cmake.in                            ########

get_filename_component(PACKAGE_PREFIX_DIR "${CMAKE_CURRENT_LIST_DIR}/../../../" ABSOLUTE)

####################################################################################

#### Expanded from @PACKAGE_DEPENDENCIES@ by install_basic_package_files() ####

include(CMakeFindDependencyMacro)
find_dependency(Boost)
find_dependency(matioCpp)
find_dependency(Threads)

###############################################################################

include("${CMAKE_CURRENT_LIST_DIR}/YARP_telemetryTargets.cmake")
```

Can you edit it like 
```cmake 
set(YARP_telemetry_VERSION 3.5.100+20211203.5+gitb7a6221dd+dirty)

####### Expanded from @PACKAGE_INIT@ by configure_package_config_file() #######
####### Any changes to this file will be overwritten by the next CMake run ####
####### The input file was YARP_telemetryConfig.cmake.in                            ########

get_filename_component(PACKAGE_PREFIX_DIR "${CMAKE_CURRENT_LIST_DIR}/../../../" ABSOLUTE)

####################################################################################

#### Expanded from @PACKAGE_DEPENDENCIES@ by install_basic_package_files() ####

include(CMakeFindDependencyMacro)
message("1--> CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}")
find_dependency(Boost)
message("2--> CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}")
find_dependency(matioCpp)
message("3--> CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}")
find_dependency(Threads)
message("4--> CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}")

###############################################################################

include("${CMAKE_CURRENT_LIST_DIR}/YARP_telemetryTargets.cmake")
```

S-Dafarra, 2021-12-17:
Hold on, maybe I got it. I think the problem might be that ``matioCpp`` is both a dependency of ``YARP_telemetry`` and of ``blf``.

GiulioRomualdi, 2021-12-17:
Just to be sure I create a brand new project
```cmake
### CMakeLists.txt

cmake_minimum_required(VERSION 3.16)
project(example LANGUAGES CXX)

list(APPEND CMAKE_MODULE_PATH ${CMAKE_CURRENT_SOURCE_DIR}/cmake)
message("--> CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}")
find_package(YARP COMPONENTS telemetry)
message("--> CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}")
```

I modified the `YARP_telemetryConfig.cmake` as follows
```cmake
set(YARP_telemetry_VERSION 3.5.0+101-20211025.9+git0baa202b9)

####### Expanded from @PACKAGE_INIT@ by configure_package_config_file() #######
####### Any changes to this file will be overwritten by the next CMake run ####
####### The input file was YARP_telemetryConfig.cmake.in                            ########

get_filename_component(PACKAGE_PREFIX_DIR "${CMAKE_CURRENT_LIST_DIR}/../../../" ABSOLUTE)

####################################################################################

#### Expanded from @PACKAGE_DEPENDENCIES@ by install_basic_package_files() ####

include(CMakeFindDependencyMacro)
find_dependency(Boost)
message("1--> CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}")
find_dependency(matioCpp)
message("2--> CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}")
find_dependency(Threads)
message("3--> CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}")
find_dependency(YARP)
message("4--> CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}")

###############################################################################

include("${CMAKE_CURRENT_LIST_DIR}/YARP_telemetryTargets.cmake")
message("5--> CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}")
```

This is the outcome of the test
```
-- The CXX compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0")  
1--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
-- Found MATIO: /usr/include  
2--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
-- Looking for C++ include pthread.h
-- Looking for C++ include pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
3--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
-- Found YARP: /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.5.0+101-20211025.9+git0baa202b9")
4--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
5--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
-- Found YARP: /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.5.0+101-20211025.9+git0baa202b9")
--> CMAKE_MODULE_PATH 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gromualdi/robot-code/test_telemetry/build
```

S-Dafarra, 2021-12-17:
I guess that's printed when you look for `yarp` (it is a dependency of telemetry). So it seems that in this case the module path is not erased. What if you also add `find_package(matioCpp)`? To you test?

GiulioRomualdi, 2021-12-17:
> So it seems that in this case the module path is not erased.

It's actually erased, please notice this
```
(found version "3.5.0+101-20211025.9+git0baa202b9")
--> CMAKE_MODULE_PATH 
-- Configuring done
```

S-Dafarra, 2021-12-18:
Oh, right, sorry I did not notice it!

What if you use
`find_package(YARP_telemetry)` instead of `find_package(YARP COMPONENTS telemetry)`?

GiulioRomualdi, 2021-12-18:
Bingo @S-Dafarra. In this case is not erased 
```
-- The CXX compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0")  
1--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
-- Found MATIO: /usr/include  
2--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
-- Looking for C++ include pthread.h
-- Looking for C++ include pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
3--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
-- Found YARP: /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.5.0+101-20211025.9+git0baa202b9")
4--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
5--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gromualdi/robot-code/test_telemetry/build
```

GiulioRomualdi, 2021-12-18:
I notice one thing. If telemetry is found using:
1. `find_package(YARP_telemetry)` the following things happen:
   - The `CMAKE_MODULE_PATH` is **not** erased
   - `Found YARP` is **not** printed after `5--> CMAKE_MODULE_PATH`
      ```
      5--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
      CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
      -- Configuring done
      ```
2. `find_package(YARP COMPONENTS telemetry)` the following things happen:
   - The `CMAKE_MODULE_PATH` is erased
   - `Found YARP` is printed after `5--> CMAKE_MODULE_PATH`
      ```
      5--> CMAKE_MODULE_PATH /home/gromualdi/robot-code/test_telemetry/cmake
      -- Found YARP: /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.5.0+101-20211025.9+git0baa202b9")
      CMAKE_MODULE_PATH 
      -- Configuring done
      ```
So there seems that one `find_package(YARP COMPONENTS telemetry)` is called `yarp` is searched once more and this erases the `CMAKE_MODULE_PATH`.

What's happening under the hood? which are the functions called when `find_package(YARP COMPONENTS telemetry)` is used that are not called with `find_package(YARP_telemetry)`?

S-Dafarra, 2021-12-18:
I guess it depends on the YARPConfig file. It should be up to the project to look for the components (https://stackoverflow.com/questions/54702582/how-to-configure-project-with-components-in-cmake)

GiulioRomualdi, 2021-12-18:
Just to be complete if I do 
```cmake
find_package(YARP COMPONENTS dev)
```
The `CMAKE_MODULE_PATH` is **not** erased

Nicogene, 2021-12-20:
Thanks @GiulioRomualdi and @S-Dafarra for the nice analysis 👍🏻 

@drdanz any idea what is the difference between this component instead `dev` for example? Maybe the fact that is external is handled in a different way?

drdanz, 2021-12-23:
I don't know, but `AddInstallRPATHSupport` is a YCM module, you should add `find_package(YCM)` before using the module.

S-Dafarra, 2021-12-23:
> I don't know, but `AddInstallRPATHSupport` is a YCM module, you should add `find_package(YCM)` before using the module.

Independently from the specific module (that in this case was copied in the ``cmake`` folder as in https://github.com/robotology/how-to-export-cpp-library/tree/afb21efb655e7b1cc11636c3d42aad9e02fb626f/cmake), the real issue is that the ``CMAKE_MODULE_PATH`` variable gets cleared somehow.

drdanz, 2021-12-23:
`YARPConfig.cmake` makes a backup of the `CMAKE_MODULE_PATH` before running, and resets it at the end.
Perhaps find_package is called recursively, and unsets the backup variable.

```cmake
# Save the variables, that it should not be modified by find_package(YARP), but
# we need to modify it in order to find all the dependencies.
set(_CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH})
set(_YCM_FOUND ${YCM_FOUND})
# [...]
# Restore the original value of the variables
set(CMAKE_MODULE_PATH ${_CMAKE_MODULE_PATH})
set(YCM_FOUND ${_YCM_FOUND})
unset(_CMAKE_MODULE_PATH)
unset(_YCM_FOUND)
```

Perhaps called recursively, therefore the second time the variable is restored, it is unset, and therefore empty.

Probably YARP_telemetryConfig.cmake calls `find_dependency(YARP COMPONENTS <...>)` instead of calling `find_dependency(YARP_<...>` like the other internal components

closed 2023-10-27 by Nicogene