robotology/robometry #156: Investigate the possibility to store the name for each element of a buffer stored by the telemetry

GiulioRomualdi, 2022-01-25:
It would be really nice to add the possibility to give a name of each element stored by a channel

Currently, the info associated with a channel is defined by a `ChannelInfo`

https://github.com/robotology/yarp-telemetry/blob/0646ef0d6c099b5a3c6291a0f7d535c46a7462e8/src/libYARP_telemetry/src/yarp/telemetry/experimental/BufferConfig.h#L24 

```cpp
yarp::telemetry::experimental::ChannelInfo channel{ "one", {1,1} };
```

One of the possibility is to use a struct to define the `ChannelInfo`
```cpp
struct ChannelInfo {
   std::string name;
   dimensions_t dimensions;
   std::vector<std::string> elements_names
}
```

To keep back-compatibility with the existing code the `elements_names` can be set as an optional parameter.

For the time being, telemetry has been used only with vectors but in theory, it can consider n-dimensional tensors. Indeed  `dimensions_t` is a `std::vector<size_t>` so it may happen that a `ChannelInfo` has the following dimensions
```cpp
dimensions = {2, 4, 1, 4}
```
In this case I have no idea how we can store the element names.

GiulioRomualdi, 2022-01-25:
cc @S-Dafarra @Nicogene @traversaro @AlexAntn

S-Dafarra, 2022-01-25:
> For the time being, telemetry has been used only with vectors but in theory, it can consider n-dimensional tensors. Indeed `dimensions_t` is a `std::vector<size_t>` so it may happen that a `ChannelInfo` has the following dimensions
> 
> ```c++
> dimensions = {2, 4, 1, 4}
> ```
> 
> In this case I have no idea how we can store the element names.

I think that it is ok to use a vector in any case. I think it should be enough to be clear about the convention to use. For example, the arrays in ``matioCpp`` are stored in column-major order, so you could add a label for each element in column-major order. In this case, it may come handy an utility that automatically resizes the label vector, and that can set a specified label given the indices (similar to https://github.com/ami-iit/matio-cpp/blob/d8a450ea7c3289ca78ea5111aea0f451a8b19b74/include/matioCpp/MultiDimensionalArray.h#L124-L144)