robotology/robometry #172: Discussion: new names for the repository and for the library

Nicogene, 2022-05-04:
This issue is for gathering the name proposal for the new name of the repository, library, namespace.

cc @GiulioRomualdi

Nicogene, 2022-05-04:
@Nicogene proposal:
- Repo : `robot-telemetry`
- Library: `robottelemetry`
- Namespace: `robottelemetry::`

S-Dafarra, 2022-05-04:
@S-Dafarra proposal:
- Repo: ``robometry``
- Library: ``robometry``
- Namespace: ``robometry::``

S-Dafarra, 2022-05-04:
@S-Dafarra proposal 2:
- Repo: ``robotemetry``
- Library: ``robotemetry``
- Namespace: ``robotemetry::``

Nicogene, 2022-05-04:
> @S-Dafarra proposal:
> 
>     * Repo: `robometry`
> 
>     * Library: `robometry`
> 
>     * Namespace: `robometry::`

I am in favour of this since it is the only name not already present in GH -> https://github.com/search?q=robometry

Nicogene, 2022-05-05:
@vtikha proposal:
- Repo : `robometrics`
- Library: `robometrics`
- Namespace: `robometrics::`

Repos with same name: https://github.com/search?q=robometrics

closed 2022-05-06 by Nicogene