robotology/robometry #174: Compilation error when BufferManager is used without template and is member of a class

Nicogene, 2022-05-10:
I committed [here](https://github.com/robotology/robometry/commit/fac3227c47aefc60e89442ee75f1365ead6d8057) a test that does not compile with this error:
```
/home/ngenesio/robotology/yarp-telemetry/test/BufferManagerTest.cpp:398:13: error: invalid use of template-name ‘robometry::BufferManager’ without an argument list
  398 |             robometry::BufferManager bm;
      |             ^~~~~~~~~
make[2]: *** [test/CMakeFiles/bufferManager_test.dir/build.make:76: test/CMakeFiles/bufferManager_test.dir/BufferManagerTest.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:586: test/CMakeFiles/bufferManager_test.dir/all] Error 2

```

With `BufferManager<int>` or `BufferManager<>` it compiles without problem.

Note that if it is not nested in a struct or class `BufferManager bm;` is a legit declaration:

https://github.com/robotology/robometry/blob/14310e9a80c75eba987d647f02a91afe9f0c0512/test/BufferManagerTest.cpp#L360

This issue affects also releases `v0.5.0` and `v0.5.1`

cc @traversaro @GiulioRomualdi @S-Dafarra

Nicogene, 2022-05-10:
You can see the failure also in the CI: https://github.com/robotology/robometry/runs/6371615035?check_suite_focus=true

S-Dafarra, 2022-05-10:
I am working on a PR

closed 2022-05-11 by Nicogene